betaflight/betaflight #400: Could you consider porting new LED features from clean flight?

skaman82, 2016-05-16:
CF included some new LED features lately. It would be really AWSOME to have this in betaflight.

Especially:
• New LED Beacon mode (Blink)
• Configurable armed/disarmed and other hardcoded colors.

Blink: https://github.com/cleanflight/cleanflight/pull/2101
Configurable colors: https://github.com/cleanflight/cleanflight/pull/2053
Cleanup (currently ongoing): https://github.com/cleanflight/cleanflight/pull/2139

Not yet in CF but would be really great: https://github.com/cleanflight/cleanflight/pull/733

jaapp, 2016-05-21:
This is not an issue

skaman82, 2016-05-21:
@jaapp i know that, this is a feature suggestion

skaman82, 2016-06-17:
See also https://github.com/betaflight/betaflight-configurator/issues/8 for Configurator changes

ozzymuppet, 2016-06-18:
Agreed. I'm surprised this isn't already implemented? 

The default arming colors don't make sense to me, so I'd love the ability to change them.

gaelj, 2016-06-22:
Any chance the parameter group feature of cleanflight will be ported here first ?

gaelj, 2016-06-25:
Working on it.

gaelj, 2016-07-18:
@skaman82 @ozzymuppet : done !

closed 2016-07-18 by skaman82