betaflight/betaflight #401: cc3d vbat pin possibly fried

artikot, 2016-05-16:
Hi,
I think I fried my vbat input pin7 on cc3d, the software isn't reading from it any more. Possibly voltage divider resistors too low - 2.2k and 9k. Using 3 and 4s.

Is there a possibility to use different pin? I don't use current monitoring.

Thanks.

readerror67, 2016-05-16:
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/target/CC3D/target.h#L99-L105

Flipflop, compile

artikot, 2016-05-16:
great, thanks 
I will have to dust off my crosscompiling skills, lol

artikot, 2016-05-18:
It looks like all my adc's are fried, getting nothing from either current monitor, rssi or vbat pins (5,6,7)
Measured my voltage divider output and with a full 3s batt its reading 2.3V.

Anyhow here is the firmware with vbat and current monitoring pins swapped.

[betaflight_2.7.1_CC3D.hex.zip](https://github.com/borisbstyle/betaflight/files/270286/betaflight_2.7.1_CC3D.hex.zip)

borisbstyle, 2016-05-19:
Sorry I am going to close this issue to prevent cluttering. Doesnt seem like firmware issue

closed 2016-05-19 by borisbstyle