betaflight/betaflight #404: BLheli Passthrough Littlebee SPF3 doesn´t work

robschii, 2016-05-18:
Hello since 2 days i have a problem with passthrough in blhelisuite.

last week is programmed my new esc´s via cleanflight mode in blhelisuite.

my littlebees comes with an older blheli firmware so i had flashed it to the last one. then i learned the esc with cleanflight configurator at the motor tab.

evrythings works. i could fly

but yesterday i want to check something in blhelisuite. i can connect over the spf3 to the esc but he can not read the setting and its find the esc without any name an firmware version.

after press check, there are no numbers and when i should connect the battery it not recognize that the battery is connected. after 15 seconds it show me a failure that i have to check the data cabel or power.

last week it was function and now it is broken.

maybe you can helb me.

Arakon, 2016-05-18:
Works flawlessly here. Make sure the right interface is selected and that you are using the latest suite, the older suite versions can not connect to the new interface code.

robschii, 2016-05-18:
i have the betaflight 2.7.0  or 2.7.1 testes and the last suite version.  could it be a connected OSD to uart

SoulPictures, 2016-05-18:
Yes  it could, if the OSD is not powered using 5V but powered with LiPo and is connected to the UART you use for USB connection, this could be the error

robschii, 2016-05-18:
ok i will test ist today, i read this in rcgroups forum in a comment

borisbstyle, 2016-05-18:
I tested and it works

closed 2016-05-18 by borisbstyle