betaflight/betaflight #405: 4S Battery Voltage Monitoring - Permanent Beep?

RipperDrone, 2016-05-18:
Hello Boris & Team:

Since having flashed up to v2.7.1, my SP3RacingF3 beeper sounds a constant alarm after flying some 1 minute from fully charged. Before, it's been the expected 6 mins flight time.

FrSky telemetry shows regular 17.6V, so does cleanflight when linking via USB when quad is still powered.

vbat_warning_cell_voltage = 35
frsky_vfas_cell_voltage = OFF

so there should be no warning above 3.5V/cell, and composed battery voltage should be transmitted over FrSky telemetry.

If I switch bat monitoring off, beeps are gone, so it definitely has to with battery voltage scaling?

SoulPictures, 2016-05-18:
Seems like you fly HV Bat, have you configured "Maximum Cell Voltage" correctly?

[image]

RipperDrone, 2016-05-18:
SoulPictures - thanks for your quick reply. Yes, I have set Vmax to 43 = 4.3V. 

Think I found the solution in the meantime, but have not completely got my head around it:

After updating to 2.7.1, the Vbat scaling factor seems to have been overwritten, even though I restored from a previously saved CLI file. Used to be 98, was now back to 110 (=default?). So triggered the beeper due to Vmax exceeded? After I manually set it back to 98, beeper stopped...

closed 2016-05-18 by RipperDrone

SoulPictures, 2016-05-18:
If 17.6 V total is correct for you, 4.4V per Cell you could also leave the Scalling as usual and adjust the cell max voltage to 4.5V then your FC should correctly detect your Cellcount, but you should also adjust the warning levels

RipperDrone, 2016-05-19:
My fault, I misinterpreted the 17.6V as correct voltage, but I just have std LiPos with 4.3V Umax. This lead me to the solution: The scaling factor has been overwritten by flashing v2.7.1 and for some reason, restoring the former settings by a CLI dump has not restored the old scaling factor (old=correct s.f. was 99, default after flashing to 2.7.1 and CLI restore was still 110). Thanks for your help - it's resolved! I did a test flight today, no more beeps.

dzenbot, 2017-11-19:
FYI, this happened to after updating to `3.2.1` from `3.2`. I was getting a warning beeper for some standard lipo batteries, but not for other ones with the same battery brand, with identical cell voltage levels. Doing `set vbat_scale = 99` fix it for me, and the OSD now displays a correct battery total voltage as well.

I'm using a Hobbywing Omnibus F4 FC.