betaflight/betaflight #406: LED_STRIP stopped working after battery swap

kaziko86, 2016-05-19:
Not sure if this is a betaflight issue, but I had a hard time getting the LED_STRIP feature to stay enabled in the GUI, ended up having to use CLI commands to enable to the feature and I got everything working fine. Flew two packs with no issues, put third pack on and the led strip at the bottom of my craft will not turn on, I tried reflashing cleanflight then betaflight etc, can't get them to come back on again, checked connections and everything is secure

KiteAnton, 2016-05-19:
You dont supply any info on your setup, but you know that LED_STRIP only is possible on PPM/SBus (or similar) setups .. ? I.e. if you reflash and still have PWM for RC input enabled then it is not possible to enable LED_STRIP.

kaziko86, 2016-05-19:
Sorry my setup is:
Emax Nighthawk X5 Frame
Emax RS2205 2300kv
Naze32 acro rev5 w/betaflight 2.7.1
Littlebee 20a w/blheli 14.5
FrSky d4rii rx

I have a wire running from the rbg pad on the pcb (has built in led strip) to pin 5 on the naze, just under the three wires heading to my receiver. I had managed to get them working, that isnt the issue, the issue is they stopped working after switching to a new battery

closed 2016-05-19 by borisbstyle

digitalomni, 2016-07-09:
Having the same problem... LED was working and then all the sudden it wasn't..