betaflight/betaflight #407: 2.7.0 mid-high throttle cut out

juanamo28, 2016-05-19:
having issues with running 2.7.0 and MS14.5

 naze32 and LB20A

 Hovers fine but when I go mid to high throttle the quad drops out of the sky and goes into failsafe mode... no esc tone and wont re arm until I replug the lipo. I tried another RX and the new one still has this issue. Tried it on looptimes ranging from 1000 - 250. I'm at a loss here on what to do. CPU is at 24% at 250 looptime so that cant be it. 
 Has anyone else had issues with this firmware on a naze32?

 I reverted back to BF 2.4.0 and blheli 14.4 and it all works fine...

borisbstyle, 2016-05-19:
just downgrade to 14.4 and use 2.7.1.

There is a known issue with littlebees and 14.5. Haa nothing to do with
betaflight

juanamo28, 2016-05-20:
I done as you said. Went to 2.7.1 blheli 14.4, hovered for a while then tried to flip and it disarms and splatters to the ground. So to me thats not an esc problem? What do you think?

borisbstyle, 2016-05-20:
looks like it or ESC min_throttle needs to go higher.

Increase min_throttle by 30 points

juanamo28, 2016-05-20:
Motors spin at 1020 and my min is already at 1050. Shall i go higher then that?

KiteAnton, 2016-06-17:
Is this still active, also in 14.6 and 2.8.1 ? 
Else we should close this issue.

closed 2016-06-23 by KiteAnton