betaflight/betaflight #409: smarport telemetry output adjustmen-t feature request

tekmunkifpv, 2016-05-20:
i am putting together an opentx configuration package/telemetry screen/soundpack (along with the cli rx values to accompany the package), that allows the inflight adjustments to be used to their fullest- to accompany the configuration i would really like to see the tune-able values on the telemetry screen-  which means being able to select the following outputs as data sources:

PID values for pitch roll yaw
expo rate
rc rate
pitch/roll rate
yaw rate
selected rateprofile (if functional)

I do realize there's likely a logical limit to the # of variables sent-  but it would be great to select the values i want from a list rather than sending the unused outputs other than vfas and the rx rssi

ultimately i envision being able to tune the quad along with seeing the tuned values immediately on the radio without needing a bluetooth module or configurator to verify values/references.

borisbstyle, 2016-05-20:
its much easier with OTG cable. No need for bluetooth.

When tuning on switch / button you dont really have to know the values.

tekmunkifpv, 2016-05-20:
i meant bluetooth to get the values in-the-air.   I do usually use an OTG in the field-  but if I could just look at my radio to see my current values-  that's a really really nice feature to have, imo.   

If you're out flying,  and something is off- you can easily check the values and adjust.   As well,  when tuning, ideally you want your 'p' as high as possible and 'd' as low as possible, so you'll know if you're getting out of spec because d can affect the 'feel'  ---   the functional use here is minimal as ultimately blackbox is the fine-tuned method, but I feel the feedback options would be ideal-  especially if inflight tuning you can actually see the value changing,  then you know the input is working.

It's a 'nice to have' feature- not a need-   Not to mention seeing the rates/expo relative to one another as you fly is a great way to find an expo/rate combo that works for your flying.

KiteAnton, 2016-06-17:
@tekmunkifpv can you (or anyone else) make a PR for this and then it can be reviewed/discussed again.

KiteAnton, 2016-06-27:
@micworg has something started for this, could you wrap it up to a PR ?

micworg, 2016-06-27:
Look here: http://www.rcgroups.com/forums/showthread.php?t=2681167 .. the code is very quick and dirty and just my personal solution. I will try to make the code more general and elegant ..

KiteAnton, 2016-09-05:
This one is fixed by #1081, closing.

PR #1081 (2016-08-31): Added PID values over smartport telemetry
PID/rate values over smartport telemetry has been requested several times, e.g. in #409 

This change adds P, I, D and rates for roll, pitch and yaw in Smartport variable Tmp2. 
Several values are sent together as an uint32, thus receiver need to bitmask values. Can be done with a simple LUA script similar to https://github.com/KiteAnton/Taranis/blob/master/tuning.lua

[image]

Fixes #409
closed 2016-09-05 by KiteAnton