betaflight/betaflight #410: BLHeli GUI

hoggdoc, 2016-05-20:
The BLHeli Suite GUI for pass through from Betaflight doesn't work anymore.  Reverted back to 2.4 everything is fine.

Is there a fix?

borisbstyle, 2016-05-20:
you need newesf blhelisuite ;)

closed 2016-05-24 by borisbstyle

hoggdoc, 2016-05-24:
Great thanks Boris.  Do you know if there are plans to port BLheli Suite to OSX?  It's a real pain for us regular Mac users to go to Windows for this app....   Wonder if it could be done in Chrome like configurator for Cleanflight and Betaflight?

Also I have searched and found the BitDump page for BLHeli, but don't see a download for the latest Suite, can you show me the link?

Thanks for all your tireless work on the Cleanfilght/Betaflight platform....

Wayne aka hoggdoc

deepimpactio, 2016-05-24:
@hoggdoc this is a Betaflight thread. You can ask that in BLHeli Github