betaflight/betaflight #413: Receiver not working ppm with cc3d

cerme0, 2016-05-21:
Hi guys. I have just updated my cc3d to the latest beaflight firmware. everything ok flashing but I don't get any response to the quad from my radio. Got a flysky i6 with ias6b receiver, 3 wires connection to the cc3d receiver port with black red and white on the left.
rx_ppm settend on receiver mode in the configurator, but no receivend nothing in receiver tab.
Not using oneshot

any help?

Arakon, 2016-05-23:
PPM pin changed to pin 8 on that plug. Just as the release notes and wiki say.

LeChuck71, 2016-06-02:
Weeks ago I had the same problem and moving ppm input to pin 8 did solve it - big surprise here.

But I think it has to be more clear that the pin changed. A lot of user came from cleanflight and there the pin stays on #3. The configurator is no big help here cause its made for cleanflight.

Maybe the wiki needs a dedicated "1. Time Setup" section?

KiteAnton, 2016-06-23:
Old issue and solution given already by Arakon, closing.

closed 2016-06-23 by KiteAnton