betaflight/betaflight #415: S.port telemetry not working with Naze32 rev6/XSR

Snapsh00t3r, 2016-05-21:
Updated from 2.6.1 to 2.7.1 and now seem unable to get telemetry to the Taranis.

Physical connections correct and verified.
Telemetry_inversion=on, running on softserial 1 (smartport telemetry enable, automatic). Tried deleting all sensors on Taranis and do new search. Nothing, even after several reboots.

digifaps, 2016-05-23:
I have trouble getting my telemetry to work on LUX. will report back after some fidling

oliranks, 2016-05-23:
im having the same issue on my sp3 evo

borisbstyle, 2016-05-24:
It works here on any board I have. It also works on my Lux board.

I dont have SP3evo

closed 2016-05-24 by borisbstyle

Snapsh00t3r, 2016-06-07:
Still not working on 2.7.1 or 2.8 with Naze32 rev6 and FrSky XSR unfortunately. Works fine when going back to 2.6.1.