betaflight/betaflight #418: 2.7.1 is "floating" on the same pids as used on 2.6.1

voroshkov, 2016-05-23:
Drone with Naze32 FC was kind of "locked" in the air with default PIDs of 2.6.1, when hovering. Same PIDs on 2.7.1 make the drone quite "floaty". 
I'm using all default settings, except PIDs and looptime = 500.

Doon44, 2016-05-23:
Me too on Xracer.  Will be interested in there response

Arakon, 2016-05-23:
Filter settings changed, may contribute.
I kept my PIDs from 2.6.1 and the copter feels nice and locked in on 2.7 as well.

deepimpactio, 2016-05-23:
Very seldom is retuning not required.

Doon44, 2016-05-23:
I have re-tuned the pants off mine in both rewrite and luxfloat and just can't get it locked in like 2.6.1?

borisbstyle, 2016-05-23:
Did you read the release notes?

There are also filter settings for 2.6.1.

If you fly level mode you can even use gyro_lpf = 188hz as well.

voroshkov, 2016-05-23:
@borisbstyle, do you mean it's all because of the filter settings? Should changing filters to values from 2.6.1 return that "locked in" feeling? 
Thanks for your reply!

Doon44, 2016-05-23:
Straight rate mode, what can be turned off is off

borisbstyle, 2016-05-23:
Yes if you have a noisy setup you need to lower the filter settings or work on your setup to make it less noisy and enjoy more locked in feel.

Try lowering D and retune for example. Might help

Doon44, 2016-05-23:
OK thanks I'll give it a go

voroshkov, 2016-05-24:
@borisbstyle Lowering D fixed the issue in my case.
Thank you, Boris!

borisbstyle, 2016-05-24:
I will close this.

Dont use more D than you need. D can cause most problems ;)

closed 2016-05-24 by borisbstyle