betaflight/betaflight #420: Accelerometer reading half in all axis on skyline32 + osd

nosredna000, 2016-05-25:
Hi,  
Just popped a naze board thought I would give a skyline a try.  Everything looks good however in the sensors tab the accelerometers are reading 0.5g in z,  put it on its side and .5g x etc. any ideas?  I've calibrated the accelerometers several times.

nosredna000, 2016-05-28:
Update.  I cant get this controller flying the way it is. Went back to cleanflight and acclerometers all read proper  1 g in z etc.    IF there is a update to fix this please let me know.  I'm using NAZE in both cleanflight and betaflight as target.  Board is a Skyline32 + OSD

mikeller, 2016-05-29:
Hi @nosredna000.

> Betaflight 2.5.0
￼>  borisbstyle released this on 4 Mar · 235 commits to betaflight since this release
> [...]
> Increased Accelerometer range from 8 to 16G

That is where the halving of the accelerometer readout in the cleanflight configurator comes from, as the configurator is not aware that the readout from the accelerometer has changed in scale.
This has no influence on the in-flight behaviour of the craft though, so it is not related to the problems you are having.
Without a description of the problems, there is no way of telling what could be the cause.

martinbudden, 2016-07-09:
@borisbstyle , is this a bug in betaflight, or the configurator?

borisbstyle, 2016-07-09:
Configurator. Betaflight has differenc acc scaling than cleanflight for the
acc.

I believe cleanflight has acc set to 8G max and betaflight to 16G

martinbudden, 2016-07-10:
@borisbstyle , the issue is that the scaling for the acc values passed by the MSP wasn't also changed.

PR https://github.com/betaflight/betaflight/pull/714 fixes this.

martinbudden, 2016-07-10:
Fixed in PR https://github.com/betaflight/betaflight/pull/714

closed 2016-07-10 by martinbudden

borisbstyle, 2016-07-10:
Btw acc scaling should be a global parameter used everywhere

martinbudden, 2016-07-10:
Yes, I know. But since we are near to v3.0 I just did a quick fix to solve the immediate problem.