betaflight/betaflight #422: configurator question

skaman82, 2016-05-25:
@borisbstyle I have been asked twice already to create a betaflight configurator. Please give me a short feedback if this is in your interest. Somebody has to maintain the app when I am through with the design changes...

If you OK with this, I would change the colors and logos as a start. 
For that I need your betaflight logo in a vector format.

RC-Slater, 2016-05-25:
Is it really needed?  I know there are unique CLI commands that don't exist in CF, but these change so often in BF anyway that it seems like a chore to maintain a separate configurator.. Also increases the barrier and learning curve for people moving from CF to BF...

skaman82, 2016-05-25:
@RC-Slater I guess people are asking because of the development speed of the CF configurator. There is almost no work being done on that comparing to the firmware an especially comparing to the development speed of betaflight. I personally have no real opinion on a separate betaflight configurator. If Boris says no, that absolutely fine with me.

borisbstyle, 2016-05-25:
Hi Albert,

Yeah a configurator would be nice to have honestly.
I introduced so many new different and experimental parameters what can be
confusing to users.

I will maintain it further once it is done.

I will also soon add some extra MSP commands betaflight specific what can
be added into configurator.

Regarding the logo. Someone else on the forum made it and I will have to
check if it is a vector?

Thanks in advance.

nathantsoi, 2016-05-25:
i think this is a great idea! i previously forked the configurator to add some exciting new features (not pushed yet... and i am glad other people think this is valuable!

i have created this repo with the basic changes (name and removed user tracking) https://github.com/airfield/betaflight-toolbox and gave @borisbstyle and @skaman82 admin rights

happy to add anyone else who wants to work on this!

nathantsoi, 2016-05-25:
one more thought @borisbstyle. i was at Maker Faire this weekend telling everyone who doesn't already use BetaFlight (most of the pros already do), that they should use BetaFlight because it is **AWESOME**!.... and i think the name "beta" might be a little scary to folks... maybe this is what we want? maybe not?

i think that since the firmware so solid (and not buggy like "beta" firmware), at some point it may be worth "re-branding" to some name that does not have "Beta" in the name :)

i bring this up now, b/c if @skaman82 is going to re-skin the configuration tool, maybe we want to do that at the same time?

deepimpactio, 2016-05-25:
.....and Awesomeflight was born

borisbstyle, 2016-05-25:
Yes nathan.....I was thinking that too just the other day.

It isnt anymore like it was initially to test cleanflight Pull Requests.

It is indeed good to change the name immediatelly.

Any suggestions?

skaman82, 2016-05-25:
thanks @borisbstyle I will work on that in the next week or two when I have time. Just finished the new INAV Configurator - moving now to the next one… (just keep them coming ;-) )

borisbstyle, 2016-05-25:
haha...you are doing a great job man!

KiteAnton, 2016-05-25:
Yes this is good. I have thought about this many times since development of cleanflight configurator seems to be going really slow. Made a [PR](https://github.com/cleanflight/cleanflight-configurator/pull/354) to fix the buffer issue for text dumps March 12th, and still no response.

mikeller, 2016-05-25:
On Thu, May 26, 2016 at 7:15 AM, borisbstyle <notifications@github.com> wrote:
> It is indeed good to change the name immediatelly.
> Any suggestions?

betterflight? ;-)

voyn238, 2016-05-25:
BorisFlight!

skaman82, 2016-05-25:
It doesn't necessarily have to include FLIGHT at the end. Or does it?

@KiteAnton I would merge your PR into CF Configurator, but I mostly merge only GUI related stuff. So Dominic has to do it, or at least comment first.

borisbstyle, 2016-05-25:
I will try to do some configurator work soon...lol.
Oh man I hate that so much :(

mikeller, 2016-05-25:
On Thu, May 26, 2016 at 7:48 AM, borisbstyle <notifications@github.com> wrote:
> I will try to do some configurator work soon...lol.
> Oh man I hate that so much :(

If you let us know what you want done, others might be able to put in some work. I for my part don't mind the odd bit of JavaScript.

RC-Slater, 2016-05-25:
@skaman oh wow your work on the iNav configurator is very nice!

borisbstyle, 2016-05-25:
@mikeller 
I just want to have an advanced tuning tab with all the "betaflight" related stuff. I get so many questions about cli commands. People somehow seem to find that hard

mikeller, 2016-05-25:
@borisbstyle: Makes sense. We should probably do it in a dynamic way, so that all the advanced parameters can be enumerated by the FC in an ordered / grouped fashion, with range and explanation added, and then dynamically build the tuning tab from this. This way new parameters will be picked up without an update of the configurator.

borisbstyle, 2016-05-26:
I have no idea about the name guys.

flight something ?

skaman82, 2016-05-26:
to bad raceflight is already taken :(

deepimpactio, 2016-05-26:
RotorFlight

skaman82, 2016-05-26:
RaceOS 
DroneFlight 
PowerFlight

borisbstyle, 2016-05-26:
@skaman82 
raceflight refers to only racing. I really want to have general acro optimised firmware not only for racing, but for freestyle or any kind of mini quad fun.

skaman82, 2016-05-26:
SmoothFlight
FunFlight
CoolFlight
FreeFlight

...just thinking loud..

deepimpactio, 2016-05-26:
PeakFlight
ApexFlight
MaxFlight
RushFlight
AdrenoFlight
FluxFlight
FlowFlight

lkaino, 2016-05-26:
Since it's no longer beta it should be Finalflight

borisbstyle, 2016-05-26:
@Ikaino
Haha....finalflight sounds like it is going to crash when you flash it :D

skaman82, 2016-05-26:
Seriously, why not AcroFlight? - so it is crystal clear what it is focused on.

flounderscore, 2016-05-26:
I would think twice before changing an established "brand" name...

robschii, 2016-05-26:
Dream flight,  fly smooth fly hard.  Dream and fly

Ark0N, 2016-05-26:
Betaflight is going into the direction of a smart flight software that is doing things more automatic like dynamic P. So I would call it SmartFlight or better Intelligence Flight (IF) or Advanced Flight (AF)

deepimpactio, 2016-05-26:
EatsYourFirstBorn Flight

schmiernippel, 2016-05-26:
EpicFlight :-)

RipperDrone, 2016-05-26:
I'm still in dispute with CLI: dumping back data never works with a complete file, it always has to be fed in several slabs. Never sure which part has been loaded and which hasn't. Being able to simply save/restore the full set of CLI features instead of having to go through the duplicate Restore and CLI paste work (which itself is tedious as explained above) is getting more & more of a nuisance. Would be dope if the GUI could handle this more easily...

bagwan, 2016-05-26:
How about Alphaflight?
I think, Betalflight is the Alpha Leader in Fc Controller Software

RipperDrone, 2016-05-26:
FreeFlight gets my backing - can be a shortcut for Freeride but also having the complete freedom of setting all available CLI options in the configurator ;-)

readerror67, 2016-05-26:
http://www.rcgroups.com/forums/showthread.php?t=1548820

5 years too late on FreeFlight

le3dmax, 2016-05-26:
@skaman82
I like RaceOS

RipperDrone, 2016-05-26:
PaceFlight (then we have Base, Race & Pace ;-)

RipperDrone, 2016-05-26:
BBFlight, put your nickname initials in

timFinn, 2016-05-26:
MultiFlight
LaserFlight
ProFlight
SharkFlight

mikeller, 2016-05-26:
@RipperDrone: 

>I'm still in dispute with CLI: dumping back data never works with a complete file, it always has to be knocked down and fed in in several slabs. Never sure which part has been loaded and which hasn't. Being able to simply save/restore the full set of CLI features instead of having to go through the duplicate Restore and CLI paste work (which itself is tedious as explained above) would be big progress.

The culprit in this is profile switching. The switch puts the FC off line for a while, and the data received during this time is lost. So to avoid it, you have to break up the dump into chunks right **after** profile switches, and hit enter to send between the chunks:

First chunk:

```
...
rxfail 17 h

# profile
profile 0
```

Next chunk:

```
############################# PROFILE VALUES ####################################

set yaw_p_limit = 500
...
```

And so on.

nathantsoi, 2016-05-27:
Wow. Naming is a hard question @borisbstyle... sorry for the short novel, I've been thinking about this for a while... The last paragraph sums it up if you want to skip there

Thinking about naming w/ the *Flight suffix vs without, I think the rename from MultiWii to BaseFlight indicated a huge shift in capability. In the same way the renaming of BaseFlight to CleanFlight indicated only a small shift in capability, whereas the upgrade from CleanFlight to BetaFlight was again a huge performance improvement. I think the new name should indicate a commensurate increase in performance, therefore I suggest we drop the *Flight suffix and make the name amazing in itself :) 
  
Going back to the name in general, I usually try to name software after other things (likely from mythology) so that associations in the users' mind of the latter are conferred to the former.

I started by looking at Dutch mythology, in homage to Boris' home country, and found out about 
https://en.m.wikipedia.org/wiki/Elegast, while an awesome name, I'm not sure about the universal pronounceability of names like this? Mythical names in general may suffer from this problem. 

So I read a bit more about https://en.m.wikipedia.org/wiki/History_of_the_Netherlands and the VOC https://en.m.wikipedia.org/wiki/Dutch_East_India_Company where I found some awesome names under the notable ship list, like Vergulde Draeck, but again, this is hard to say and hard to explain. 

Then I read a bit of https://en.m.wikipedia.org/wiki/Swiss_folklore, inspired by STmicrocontroller, but didn't find anything interesting there... 

So I pivoted themes a bit and looked up the fastest ships in sci-fi https://www.fatwallet.com/blog/fastest-ship-in-the-universe/, which got me thinking, "is the improbability drive really that fast? And if it exists everywhere at once, should I be thinking less linearly?" So I went to bed. 

I woke up thinking about birds. A friend pointed out that a hummingbird has the ability to hover and, among birds, probably the most precise control of their flight. Google translate tells me that In Dutch, "hummingbird" is “Kolibrie” which seems cool, but there is already a colibri flight controller. The hawk, predator of the sky, is “Havik” in Dutch, which is really cool. Also “Alder” or eagle in German… but that seems over-used. 

Unless any of these ideas have mass appeal, my search for the ultimate name for the ultimate firmware continues... 

Ok. After dinner and a beer, I think mythology may be the way to go, but something more pronounceable like... **Pegasus**. I bet an awesome designer like @JackReis could make us a better logo than these...

[image]

[image]

I would say we should name it Pink Pegasus, but I can’t see myself saying to someone, “Are you using the Pink Pegasus firmware for your Naze?” but IMHO, “I fly Pegasus,” sounds pretty awesome... Other dude: "What you fly a Pegasus?" Me: "No, I fly Pegasus, the firmware formerly known as BetaFlight..."

If none of those ideas sound awesome to y'all, let's go with Flighty McFlightFace

RC-Slater, 2016-05-27:
[image]

:)

mikeller, 2016-05-27:
Totally Pegasus.

RipperDrone, 2016-05-27:
@mikeller Thank you for Südring some light on the CLI chunks die to profile switching - Tage me a great Heureka Moment ;-)

smiling-Jack, 2016-05-27:
**Dragon Flight** with a cool logo like this 
[image]

nathantsoi, 2016-05-27:
hum... awesome or terrible...

[image]

skaman82, 2016-05-27:
Well... I would go with terrible :)

smiling-Jack, 2016-05-27:
No its not only you ! It is terrible.

19Gerhard85, 2016-05-27:
What about PWNFlight ? ;)

And as a slogan: One FC OS to pwn them all!

skaman82, 2016-05-27:
@nathantsoi you clearly have to much free time ;-)

AndersHoglund, 2016-05-27:
I would not touch an established brand name. Just a name change with no new content is just cosmetics.
There are bigger things coming, from what I have understood.... Save the new name suggestions for that. 
/A

RipperDrone, 2016-05-27:
FlightClub (like Fight Club) - switch word order?

RipperDrone, 2016-05-27:
simply: Racer - or - Razor?

RipperDrone, 2016-05-27:
X-Flight, X stands for a quadcopter shape, and for variable tuning features (variable is x)

RipperDrone, 2016-05-27:
How about s/t more unrelated - like 

Gates
TopView
TuneMyCopter, TMC

RipperDrone, 2016-05-27:
ok, creative moments over now...

Doon44, 2016-05-27:
Phoenix flight

Marsupilami-man, 2016-05-27:
BetterFlight

[22 more replies]

closed 2016-06-13 by skaman82