betaflight/betaflight #423: An extremely serious issue. Motor may start at 1500 throttle after saving a configuration.

nickpgre, 2016-05-25:
I have a naze32 Rev6 with the battery plugged in.

This can be replicated after enabling the "Enable battery current monitoring" within the configuration screen, after saving the changes the motors start with 1500 throttle applied. The only way to make them stop if you still have your fingers attached in your arms is to switch to Motor Testing screen and disable them from there.

Extremely dangerous since there is no reason to get the props off since there was no intention to perform any motor testing.

borisbstyle, 2016-05-25:
I think you accidantaly change thingd to 3D. In 3D mode 1500 throttle is
actually where props dont spin.

nickpgre, 2016-05-25:
Sorry to say that but I can replicate this at any time by just enabling the "Battery current monitoring"
The settings I had marked (some of which are not stored) are:

1. Enable Battery voltage monitoring
2. Enabled serial-based receiver (unable to store/save)
3. Don't spin the motors when armed
4. I disabled the "Enable GPS (PPM or 3rd serial port required)"
5. Enable sonar (unable to save)

Some changes under Throttle settings, Serial receiver as I am using sbus and Battery voltage.

I have the 0.68.3 version installed on my computer and seems to be the latest version according to chrome store. It would be great if you point me out to a newer version as I would like to give it a go.

borisbstyle, 2016-05-25:
can you check if this issue is there on cleanflight as well as its a
cleanflight configurator.

Btw....changing stuff in configurator requires removing props and battery
disconnected.
Oneshot125 protocol can shortly also fire up the motors

Arakon, 2016-05-25:
Your configurator version is extremely outdated. The latest one is 1.2.2. This is quite likely the issue.
Searching on the chrome store for "cleanflight" yields the latest version here.

nickpgre, 2016-05-25:
Very happy to hear that.
I will try it ASAP. Thank you

closed 2016-05-27 by nickpgre