betaflight/betaflight #430: 2.7.1 Loss of Control After Aggressive Roll

zschiffelbein, 2016-06-01:
Flashed a new build with betaflight 2.7.1. Hovers and flies around great until any aggressive roll maneuver. As soon as the stick is centered and the roll completes the quad starts spinning out of control on the yaw access. At first I assumed it was an ESC desync issue so I upgraded BLHeli from 14.4 to 14.6 and changed the motor timing from medium to medium-high. This issue still remains. I then flashed another build that I have been flying for months without issue from betaflight 2.4.1 to 2.7.1. The same problem started on that build as well. Here is a [Blackbox Log](https://www.dropbox.com/s/k2r02on393geyz8/blackbox_log_2016-05-31_120034.TXT?dl=0) of the issue.

Build 1 - ImpulseRC Alien 6"
* SPRacing F3
* DYS XM30A (BLHeli 14.6)
* EMAX RS2205 2300KV
* DAL 6045

Build 2 - Lumenier QAV210
* Naze32 Rev6b
* DYS SN20A (BLHeli 14.1)
* EMAX RS2205 2300KV
* DAL 5045

borisbstyle, 2016-06-01:
increase min_throttle on blheli

zschiffelbein, 2016-06-01:
Bumped the min throttle to 1100 on Build 1 and the problem persisted. At 1100 the quad nearly left the ground on arm.

borisbstyle, 2016-06-01:
@zschiffelbein 
Than its something else. Nobody else actually experience this and I think its something with your ESC's. 
What escs are you using? Sometimes setting demag too high can help on some blheli ESC;s

zschiffelbein, 2016-06-01:
I thought this as well, but the fact that it just started happening on my second quad that has been flying great on betaflight 2.4.1 for months had me concerned that it was the firmware upgrade. One quad has XM30A and the other has SN20A.

borisbstyle, 2016-06-01:
SN20 have always been a problem on mine.

There must be something else you changed probably during the upgrade.
Perhaps newer blheli firmware?

Downgrade to 2.4.1 to verify better perhaps.

There hasnt been changed anything in the way how the fc sends the signal ti
tge esc for ages.

zschiffelbein, 2016-06-01:
I'd be happy to try downgrading to 2.4.1 again to test it out. The only setting I changed on that quad was the loop time to 500. No ESC settings or firmware were changed.

zschiffelbein, 2016-06-01:
Do you have a preferred version of BLHeli I should try?

borisbstyle, 2016-06-01:
14.6 or 14.4

borisbstyle, 2016-06-01:
> Just installed betaflight 2.6.1 on build 1 with no changes to the ESCs and
> the problem appears to have gone away.

Problem didnt go away, but 2.6.1 limits the power when esc shows a short
desync.
Thats a known thing.

zschiffelbein, 2016-06-01:
Interesting. So the problem has most likely been there the entire time it just was being masked by betaflight?

borisbstyle, 2016-06-01:
yup

zschiffelbein, 2016-06-01:
Thank you for your help @borisbstyle . After manually calibrating the ESC in BLHeli Suite (1100 - 2000) I was able to re-install betaflight 2.7.1. I still have a minor desync issue, but it no long yaws out of control. I am going to continue refining my BLHeli Settings.

closed 2016-06-01 by zschiffelbein