betaflight/betaflight #433: 2.8.0 RC4 crazy spin-up

RipperDrone, 2016-06-01:
Flashing RC4 and CLI loading the RC3 parameters caused my SP3RacingPro to spin all motors up to max speed immediately after throttle input raised to >0. Could only cut power to save it from doing further weird things.

Is it because overwriting with RC3 profiles causes some values being off sweet spot?

borisbstyle, 2016-06-01:
Was this in level mode?

Try RC5 and reset all settings

RipperDrone, 2016-06-01:
yes, it was level.Yep, will verify with RC5. Didn't find it yet on Jenkins.

BTW, your development branch here is awesome! I'm a big fan, and feel like even after sending a donation there can't be enough credit for your excellent work for us as a community. Thank you 1k times!

RipperDrone, 2016-06-01:
Flashed RC5, checked default parms --> all good
Re-loaded RC3 parms --> still good

Works like a charm now! Maybe your item 'fix broken level mode' was the key...

closed 2016-06-01 by RipperDrone