betaflight/betaflight #436: Motor 100% suddenly without reason CC3D Atom+Betaflight 2.7.1+DYS SN20A BLHeli 14.5

chilly24, 2016-06-02:
Dear Drone Colleaguess,

i have flashed my CC3D Atom with Betaflight 2.7.1 and when i am flying after about 30-60Sec my Motor 1 has suddenly 100% power and the drone crashes....

Hardware:
CC3D Atom
FRSKY D8R-2 plus with PPM
DYS SN20A with BLHeli 14.5.0

Here is my blackbox recording:
http://www78.zippyshare.com/v/y2LEqTLf/file.html

I got no problems before betaflight

Many thanks in adivce for help!

Kind regards

borisbstyle, 2016-06-02:
Thats your esc what desyncs.

Try betaflight 2.8. that one has a desync protection

chilly24, 2016-06-02:
how should i flash to 2.8?

When i opened Betaflight>Firmware>loaded CC3D 2.8.0.hex-->fail to load serial port
Manuel baudrate did not help and i also getting no connection now in openpilot..

LeChuck71, 2016-06-03:
Sounds like you still have the OpenPilot bootloader on your CC3D. In this case you should have used **betaflight_2.8.0_CC3D_OPBL.bin** (OPBL=OpenPIlot Boot Loader)

Connect via an FTDI adapter and flash the hex. But this way going back to OpenPilot comes with a bit more work.

chilly24, 2016-06-03:
i unpacked my cc3d atom now, will flash after work.
I am lucky that i am already having a FTDI Adapter, i think i have to connect the Bootloader PINs on the back?
Why do i have to use the betaflight_2.8.0_CC3D_OPBL.bin? When i am connecting directly via FTDI Adapter the normal betaflight_2.8.0_CC3D.hex should work?

chilly24, 2016-06-03:
ok flashed succesfully to 2.8.0 was a little bit tricky i had to set in the FTDI Adapter a manual baud rate.

@borisbstyle do i need to enable something else or runs the desync automatic?

borisbstyle, 2016-06-03:
no leave it on default first

chilly24, 2016-06-03:
thanks i will try

chilly24, 2016-06-05:
@borisbstyle 
i tested now with BF 2.8.0 but i still get the same problem with the desync again
Any ideas? I mean i could replace this esc because it is everytime the motor 4 but i have some other desnyc during hovering, see picture 2

LOG
http://www96.zippyshare.com/v/5k6Ebhwm/file.html
Screen:
https://picload.org/image/rgpgwllr/log2.png

https://picload.org/image/rgpgwwgl/log3.png

KiteAnton, 2016-06-27:
@chilly24 have you found a solution to your problem yet? If so can you comment and/or close this issue? 

Since this seems to be more related to failing hardware and not related to betaflight FW I would suggest this issue should be closed.

closed 2016-06-30 by KiteAnton