betaflight/betaflight #438: Wrong fast_pwm_protocol defaults for brushed targets in 2.8.

grokasm, 2016-06-03:
The ALIENFLIGHTF1 2.8 build causes the motors to spin up upon battery connection.
I believe this is due to having the incorrect default config for the `fast_pwm_protocol`.
There is a note of this in the wiki, but it has not been implemented as defaults for these targets. 

https://github.com/borisbstyle/betaflight/wiki/Betaflight-specific-CLI-commands

[image]

blckmn, 2016-06-14:
These have been changed in 3.0. Now you have the option of BRUSHED as a motor_pwm_protocol (along with the previous others).

Can you please confirm this issue still exists in the development branch?

borisbstyle, 2016-07-03:
Fixed in 3.0

closed 2016-07-03 by borisbstyle