betaflight/betaflight #439: Low battery buzzer goes off when plugged into USB

fahrvergnuugen, 2016-06-03:
Version 2.8, RMRC Dodo rev 3a. When plugging into USB  (no battery power) the low voltage buzzer is tripped. I have to disable feature VBAT to keep my sanity :)

A simple solution would be to ignore the battery voltage if its between 5.3 and 5.6v (USB power), unless there is a simpler way to detect when the board is plugged into USB.

borisbstyle, 2016-06-03:
try beeper USB or -USB whatever it is.

fahrvergnuugen, 2016-06-03:
Forgot to mention that I tried that. It doesn't work.

`# beeper`
`Disabled:  ON_USB`

borisbstyle, 2016-06-03:
Than that board doesnt support it unfortunatelly

borisbstyle, 2016-06-03:
Without lipo a board should not allow any voltage onto VBAT pin and Dodo
apparently does that. None of the other boards I know do that.

2016-06-03 15:46 GMT+02:00 Boris B. <dreambb1982@gmail.com>:

fahrvergnuugen, 2016-06-03:
I'll send RMRC a link to this ticket, maybe they will change it in a future revision to the board.

borisbstyle, 2016-06-03:
Look when there is a voltage on the pin it will automatically think there
is lipo connected. Having 2S lipo connected is completaly plausible for the
FC!

flounderscore, 2016-06-03:
I can confirm that issue. Possible workaround would be to disable the
beeper when VBAT goes below 6V (which would never happen in flight on 3 or
4S).
@borisbstyle Dodo has a bult-in regulator. I suspect that that causes
voltage on the vbat rail.

fahrvergnuugen, 2016-06-03:
The built in regulator is also on my Kiss FC, I imagine you will have this same problem if/when Betaflight is ported to that board. Those are the only two boards I have at the moment, but I'm surprised there aren't others that are powering the RX for convenience sake. It's nice to not have to plug in a LIPO to power the RX (makes config changes via GUI easier/safer).

borisbstyle, 2016-06-03:
Kiss FC is not a betaflight target.

Motolab Cyclone has a built in regulator too. That one does support ON_USB.
There should never be 6V on the VBAT pin! There is a voltage divider and
everything. Something is not right about the design. You can't just ignore
6V as it can be a 2S lipo!

borisbstyle, 2016-06-03:
> Another problem with USB power is that you cant flash ESC's as they will
> boot too soon
>
> 2016-06-03 16:07 GMT+02:00 Boris B. <dreambb1982@gmail.com>:

You can also simply disable all beeper notifications "beeper ALL"

2016-06-03 16:07 GMT+02:00 Boris B. <dreambb1982@gmail.com>:

fahrvergnuugen, 2016-06-06:
Something I just noticed. Regardless of whether or not the battery is plugged in, the voltage readout in Cleanflight configurator is 0v.

closed 2016-06-09 by borisbstyle