betaflight/betaflight #440: Yaw input while stick arming

timdonovanuk, 2016-06-03:
Hi,

BF 2.8.0

I have min_check = 1026 and my throttle is at 1002 when at minimum. I'm seeing weird shuddering from yaw input when trying to arm and disarm. Whenever I arm or disarm the quad twitches in the direction of yaw stick. The twitching gets more intense as I move the stick more into the arm/disarm position. 

My understanding is with min_check = 1026 and my throttle at 1002 there should be zero yaw input?

The shaking is also very weird...

I took a video: https://www.youtube.com/watch?v=LM3o2GDO9o4 (sorry its sideways, hopefully YouTube has processed its rotation by the time anyone sees this..)

borisbstyle, 2016-06-03:
Sounds like you are hitting the min check perhaps. 1020 is quite low and
easy to reach.

Try higher min check for stick arming. But even better....try switch arming
;)

timdonovanuk, 2016-06-03:
Hey Boris :)

I have min_check at 1026 though, and I've monitored the receiver in CF config - my throttle never goes above 1002 while stick arming. 

I will switch to switch arming soon thou :)

Cheers.

timdonovanuk, 2016-06-07:
It seems like something in the BF algorithms is making P gain go crazy even when yaw is under min_check. In the blackbox logs, you can't even see the yaw rc input, but here is how that weird shuddering manifests as soon as you give it a bit of yaw on the ground at 0% throttle. In this pic I've done yaw to the left a bit and then yaw to the right a bit:

[image]

borisbstyle, 2016-06-07:
> [image: 2016-06-07 22_17_58-cleanflight memoryleaks
> org_blackbox-log-viewer_]

Try beraflight 2.8

timdonovanuk, 2016-06-07:
This is on 2.8.0

borisbstyle, 2016-06-07:
https://dl.dropboxusercontent.com/u/31537757/betaflight/betaflight_2.8.1.zip

closed 2016-06-09 by borisbstyle