betaflight/betaflight #441: weird vibrations after frame change

raymonkroeze, 2016-06-03:
i have a weird problem with my new frame. i use to fly a 210 frame and then i switched the same gear over to a krieger like frame and i have stong vibrations at aprox 3/4 and up. I can't tune this vibrations out. im using rs2205 2300kv motors, littlebee 20A (and 20A pro i tried). sp racing f3 evo (and i tried the sp racing f3). I tried bf 2.6.1 & 2.7.1 and 2.8. but all give the same result. i also upgraded blheli swithed to multishot. did try 1khz to 8khz and all give the same issue. also i've noticed the fpv camera gets blurry sometimes. At first i thought it was maybe a too high yawPgain. i dropped the pgain by half and nothing has changed.
What can be causing this? everything should be working fine. and the frame is very stiff so that can't be the problem.

LeChuck71, 2016-06-06:
If the vibrations get pretty worse at a specific throttle setting it sounds a lot like "resonance".

Caused by just you're current "Mass Setup" - change some placement of components or tighten/loose a zip tie and it can be solved ... or get worse.

You can check if its caused by a singel motor by slowly raising throttle via GUI and check the accelerator readouts. Problem is that without props the resonance can be smaller or completely gone but with props its there.

raymonkroeze, 2016-06-06:
yesterday i did a lot off testing. I finally got it flying without vibrations. the problem was, i was having way too much Dgain on pitch and roll and way too high Pgain on yaw. I dropped the Dgain back to 1 and de pgains to 3. also i did the Igain in halve. from there i could start tuning and got rid off the vibrations and wobbles

closed 2016-06-06 by raymonkroeze