betaflight/betaflight #445: Can't disable oneshot in version 2.8.0 on CC3D

jregenstein, 2016-06-04:
I tried both using the GUI switch to disable oneshot, and inputing the command "feature -oneshot". After saving and resetting oneshot always went back to being enabled. Flashing betaflight 2.7.1 solved the issue. Flight controller was a CC3D (I know I should upgrade, I have a motolabs cyclone for the next build).

deepimpactio, 2016-06-04:
@jregenstein The release notes say the following: Seperated fast_pwm_protocol from ONESHOT125 feature. The feature in GUI will only be green when ONESHOT125 is actually selected. Normal traditional PWM can be now selected with "set fast_pwm_protocol = OFF"

https://github.com/borisbstyle/betaflight/releases/tag/v.2.8.0

jregenstein, 2016-06-04:
Oh, sorry, missed that bit. Thanks!

closed 2016-06-04 by borisbstyle