betaflight/betaflight #447: yaw_jump_prevention_limit --> Yaw D: Does it work?

RipperDrone, 2016-06-05:
Hello Boris and community

Can anybody confirm that yaw D setting has an influence on yaw jump prevention? I have tried all settings from 0 through 255 (max which Cleanflight GUI will allow) and held the quad in my hand while doing quick yaw blasts at 2/3 throttle. Thrust/uplift power as a reaction to sudden yaw stops virtually remained unchanged throughout the whole range of yaw D. I'm gonna test it out in the field as soon as rain/wet allows but currently I have my doubts whether or not it's factored into the software calculations correctly. If I got it right, you multiply the yaw D by 8 and subtract it from the max limit of yaw prevention, so at a setting of yaw D = 50, it corresponds to a former yaw_jump_prevention_limit of 500-50*8 = 100. BTW, what happens if I set yaw D to 255, will it cause negative jump_prevention_limit?

deepimpactio, 2016-06-05:
@RipperDrone this is in release notes for 2.8.0 if you are using that version: yaw_jump_prevention is removed and replaced by yaw D gain! The higher the D the lower the jump_prevention limit will be. 

https://github.com/borisbstyle/betaflight/releases/

deepimpactio, 2016-06-05:
@RipperDrone https://youtu.be/FrGaxDGx0fA

RipperDrone, 2016-06-05:
@deepimpactio thank you for your reply. I know that it's been replaced by yaw D, therefore I tried different yaw D settings to see the effect - but there was none! It seems the yaw D value is not making its way into the yaw prevention algorithm correctly.

deepimpactio, 2016-06-05:
Ah OK thought you might have missed it. I apologize.

RipperDrone, 2016-06-05:
No worries, maybe when you have the chance you can try setting different (high / low) yaw D values and double check if there is any effect on yaw stop as far as jump reactions?

GaryKeeble, 2016-06-06:
I took a look at the code; it is certainly in there and although it has moved in position (from mixer logic to the PID controller's), it looks like it functions the same way.

In answer to your question, the resulting yaw jump prevention limit from the calculation is constrained between 80 and 400 so negative results are protected. Also, a value of YawD set to 25 would result in exactly the same jump protection limit of 200 (400-25*8=200) that was the previous original default.

Due to the software constraint the useful working range of YawD is 0 to 40;

As a side note, it is a little confusing (for me) to think of it as YawD as it is not D-Term at all (Yaw is a PI controller only); I believe the plan is just to re-purpose the parameter on the GUI to make it easier to change (from your OSD for example).

borisbstyle, 2016-06-06:
@RipperDrone 
You increased your yaw rate by much more probably and thats why you experience more jumps probably.
With super expo there is a curve at the end and yaw gets more speed with more stick input. Even though you think you have the same yaw rate as it feels the same it is probably much faster.

Faster yaw rate = more jump

RipperDrone, 2016-06-06:
@borisbstyle Good point, yaw rate is set to 0.70 and RC expo = 0.1. In your excel tool, this shows very much the same curve as before with old rate style settings - I will try reduce yaw rate anyway. 

Thank you!

borisbstyle, 2016-06-08:
there are 2 things. Yaw jump during yaw and yaw jump on stops.

yaw D only helps on stops and rate and yaw p limit during yaw.

RipperDrone, 2016-06-08:
Yaw jumps (which by my understanding is quad popping up = burst in uplift) are still quite strong on Yaw Stops. During yaw, height is maintained quite well.

I tried playing with yaw D from 0 to 40, no perceivable change in behavior. 

Only occurs in instant yaw stops / let joystick bounce back to center. When doing smooth yaw stops, it's all fine.

borisbstyle, 2016-06-09:
Lower your yaw rate for less jump ;)

Actually the jumps are not a big deal. For snappy yaws there is always a jump depending of the copter. When you yaw hard just go zero throttle. Thats the practical case

RipperDrone, 2016-06-09:
Ok, thank you for giving direction to me. Will close this thread

closed 2016-06-09 by RipperDrone