betaflight/betaflight #449: adjrange: CLI parser does not accept adjust function numbers >20

RipperDrone, 2016-06-05:
It seems like the CLI parser does not allow for adjrange values >20 for the PID adjustment functions. Therefore, my last 6 lines (see below) would not be loaded. If I replace function values 21...26 by lower numbers, it reads all the lines fine.

On the other hand, CLI **dumps** the adjustment functions >20 correclty (like VEL P, I, D or ALT P...). Therefore, it seems CLI cannot **read** in what it has dumped itself ;-)

Solution: Parser range for function numbers needs to be extended beyond 20 to include all available functions for inflight PID tuning, like VEL, ALT, ... I assume it's just a glitch because in early versions adjustment functions were limited to a range 0...11, then extended to 20 and finally beyond 20. Looks like the parser range has not been synchronized after the last range extension.

Please include into a future build if possible.
So long, workaround is to load via GUI from the text file after CLI-loading

example:

# adjrange

adjrange 0 0 3 1000 1050 6 2
adjrange 1 0 3 1075 1125 7 2
adjrange 2 0 3 1125 1175 8 2
adjrange 3 0 3 1175 1225 9 2
adjrange 4 0 3 1225 1300 10 2
adjrange 5 0 3 1300 1375 11 2
adjrange 6 0 3 1375 1425 4 2
adjrange 7 0 3 1425 1475 5 2
adjrange 8 0 3 1475 1525 1 2
adjrange 9 0 3 1525 1575 21 2
adjrange 10 0 3 1575 1625 22 2
adjrange 11 0 3 1650 1700 23 2
adjrange 12 0 3 1700 1750 24 2
adjrange 13 0 3 1750 1800 25 2
adjrange 14 0 3 1800 1875 26 2

(latter six not effective...)

KiteAnton, 2016-06-05:
This is actually correct, there is only 20 adjustment functions defined for betaflight. 
What are you trying to adjust?

KiteAnton, 2016-06-05:
If you check here, there is only these 20 functions that is possible to tune in betaflight. 
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/io/rc_controls.h#L168-192

RipperDrone, 2016-06-06:
@KiteAnton The Cleanflight GUI has a lot more you can select: 
[image]

I would like to set ALT P and VEL P, I, D, respectively. All of them are shown in the drop-down GUI dialog, but as per your hint, the enum section does not seem to have code for it. Can it be added to make the GUI's functionality match with the BetaFlight code?

KiteAnton, 2016-06-18:
The gui and Fw is not synced with possible adjustment functions. I think there are quite few people using betaflight with the need for those inflight adjustments. 
If anyone are interested to make a PR for this then go ahead, I close this issue since there is no bug/issue with the FW.

closed 2016-06-18 by KiteAnton