betaflight/betaflight #454: Allow radio to create logic high/ low on GPIO pins

PiManASM, 2016-06-07:
Would it be possible to have a radio channel drive output a logical high/ low on the GPIO pin?

Thanks
PiMan

lunddahl, 2016-06-07:
This would probably not be very hard to accomplish, however it would be a pain in the ass to maintain compatibility with every board supported, as the CPU pins is not just floating freely on all boards.

You will also have to inspect every hardware carefully to secure that the output state is as wanted during the power up process. People will put dangerous things on those outputs eventually.

PiManASM, 2016-06-07:
What I would like to be able to do is emulate the FX799T button from my radio in order to change the channel on the VTx without access to the button. It seems like it should be possible with either a transistor or perhaps just a straight wire. 

Where I have mine mounted now it's hard to reach the button, but near the FC.

mikeller, 2016-06-07:
Hi @PiManASM.

You might be able to make this work by using a spare PWM output and a PWM to analog converter like this one: http://electronicdesign.com/analog/simple-circuit-turns-pwm-digitally-adjustable-precision-reference , and setting up a switch in your TX's mixer to output the right signal levels to drive the output to the right voltage for low / high.

KiteAnton, 2016-06-27:
If someone wants to make a PR for this go ahead, but be sure to maintain compability as mentioned above. Since it is an old issue and no one hasn't really jumped on the idea yet then I will close this for now.

closed 2016-06-27 by KiteAnton