betaflight/betaflight #457: naze32 rev5 2.8 horizon mode violent flip over

fkam18, 2016-06-07:
My setup was working fine in 2.8 rc2 but on 2.8 in particular in horizon mode any slight roll movement will flip the whole drone violently left and right and even flipping over crashing straight to the ground that sort. I tried to hover it a few inches above ground but the same. It just flips over with blades digging in the mud.

It's straight from flashing no tuning at all.

borisbstyle, 2016-06-07:
Will be fixed in 2.8.2

fkam18, 2016-06-07:
Thank you so much.

closed 2016-06-07 by fkam18

fkam18, 2016-06-10:
Oh just for your information, the 2.8 rc2 works perfectly on horizon/angle and rate  I just re-tested and am using.

mr2web, 2016-06-12:
I'm experience very high over-sensitivity in my controls when going into Horizon mode. In Acro and Air mode everything is just fine. I'm using PID crtl 1 (0 in cli) with LUX. my version of 2.8.1 rc1. I did not experience this in 2.8.0.