betaflight/betaflight #458: Extra safeguards for switch arming

TonyLeyland, 2016-06-08:
I've noticed the following when using stick arming:

1) Place throttle stick away from zero position
2) Arm the copter with switch
3) Motors do not spin
4) Bring throttle back down to zero
5) Motors start to spin !

As an extra safety feature how about the following:

At step 5 the motors do not spin. The only way to get the motors to spin would be:

a) Put throttle to zero
b) Arm the copter with switch
c) Motors start to spin

Hope that makes sense, what do you think ?

Cheers
Tony

KiteAnton, 2016-06-08:
This can be solved easy in the transmitter. 
1. Use a sticky switch in the Tx and only allow it to enable when both the physical switch is in its armed state and the throttle is at the bottom. 
2. The sticky switch should be disabled with only the switch and no conditions for the throttle value.

LeChuck71, 2016-06-08:
Hm, sorry but that does not make sense to me.
Isn't it already a safety feature that the motors don't spin up while throttle is in a none zero position and the copter ist armed via switch? Otherwise you could arm at max throttle and the copter would go off after you switched "on" ;)

I've set up my TX that all switches have to be i a default position on switching "on", so I can not mistakenly have the arm switch on "ON" position.

borisbstyle, 2016-06-09:
This is a perfect safety feature actually. Throttle must be zero to arm!

But this doesnt make sense? You arm on stick, but you describe switch?

> I've noticed the following when using **stick** arming:
> 
> 1) Place throttle stick away from zero position
> 2) **Arm the copter with switch**
> 
>

TonyLeyland, 2016-06-09:
Sorry for confusion, I tried to make it easy to read but I made a mistake since it was late. Please ignore the sentence "I've noticed the following when using **stick** arming", it should of read "I've noticed the following when using **switch** arming:"

If you now re-read my statements it should hopefully make sense and if you try what I'm saying I think you will agree that it's a bit of an issue. I was explaining to a new pilot how good switch arming was and how safe it is compared to old-school stick arming, when I got a scare since my switch was armed but when I brought my throttle down to zero the motors started spinning much to my surprise !

I like the TX feature described above and that solution I will implement. However, not everyone knows how to do this. So I'm requesting that Betaflight could be configured to do this instead of in the TX so everyone benefits.

KiteAnton, 2016-06-16:
@TonyLeyland I understand your point/concern but this seems to be of low priority to fix in FW. You (or anyone else) are always free to make a PR and then it can be discussed again, but closing this issue for now.

closed 2016-06-16 by KiteAnton