betaflight/betaflight #461: Feature Air Mode not activating properly?

mike-levenick, 2016-06-09:
Perhaps I'm just misunderstanding it, but I'm not sure the new air mode feature is working properly. 

Steps to recreate:

0) REMOVE PROPS
1) Turn on feature airmode
2) Set air mode activation level to something reasonable like 1300
3) Make sure your props are still removed
4) Arm quad (motors do not spin) 
5) Throttle up beyond air mode activation point (I have taken throttle to 80% and held it there for a few seconds just to be sure)
6) Drop throttle stick back to zero
7) Motors stop spinning
8) Grab quad and move it around to be sure the PID controller is completely disabled

---

As an aside, with feature air mode turned on, and motor stop turned OFF, I did the following:

1) Plug quad in and go to tethered logging in cleanflight
2) Arm quad, motors spin at min throttle
3) Do NOT throttle up to activate air mode
4) grab quad and move it around
5) Check logs, some motors went to max throttle trying to re-level the quad, even before reaching the air mode activation point.

---

I began digging into this because I did the following:

1) Set quad on backpack that I normally use for a launch pad
2) Put RX into bind mode
3) Put taranis into bind mode and set down
4) Unplug and plug quad back in to complete bind
5) Flip arm switch to test bind
6) Realize quad isn't arming because throttle is up
7) Zero out throttle, quad immediately arms because arm switch is still up
8) a corner of the backpack that the quad is resting on collapses, quad loses attitude, props spin up, q

mike-levenick, 2016-06-09:
After setting air mode back to a switch today, I'm relatively certain feature air mode isn't doing anything at all, and that the reason it felt like it was before was because you increased PID authority at low throttle even outside of air.

I put it back on a switch, and my zero throttle rolls were SIGNIFICANTLY faster than when I had the feature enabled.

isilcala, 2016-06-09:
I can confirm my TL130H1 with NAZE has the same behaviors. Seems Airmode is not properly activated.

borisbstyle, 2016-06-09:
any blackbox logs? Blackbox tells you if the airmode is activated or not

mike-levenick, 2016-06-09:
Unfortunately the powercube does not have a built in blackbox. I'm relatively new to logging, can I store the same info with tethered logging via USB? Otherwise I have an F3 I may be able to flash this evening that I think has BB, the quad is in pieces, but should be able to log inputs/outputs from the FC just fine.

I do have a video of flips/rolls comparing feature air mode to air mode on switch, and you can see that with switch they are faster, but that's rather subjective compared to a log I know.

Is feature air mode supposed to STAY on after it is activated at ~1300, or does it turn off every time you dip below that throttle point once it is initially activated at takeoff?

mike-levenick, 2016-06-15:
[black-box-and-CLI-dump.zip](https://github.com/betaflight/betaflight/files/317356/black-box-and-CLI-dump.zip)

Here's a zip of 2.8.1 running on F3 with:
Feature airmode ON - Motor stop ON - Airmode activation set to default 1350
Feature airmode OFF - Air mode on switch 
Feature airmode ON - Motor stop OFF - Airmode activation set to default 1350

The CLI dump is from the last test of airmode ON motor stop OFF.

Unless I am misreading the blackbox logs, airmode never comes on from feature airmode, but does come on when bound to a switch. 

This agrees with my subjective viewings of my zero throttle flips and rolls being significantly faster with airmode on a switch.

mike-levenick, 2016-06-15:
I"m also willing to do any additional testing or logging you need. The only quad I have on 2.8.x with blackbox is the F3, but can do non-blackbox tests with the powercube/calibri quad as well.

borisbstyle, 2016-06-18:
In this log
Switch-airmode-motor-stop.txt

Your throttle never went above 1350!

borisbstyle, 2016-06-18:
In all other (feature) logs air mode seems active!

mike-levenick, 2016-06-19:
@borisbstyle , the log switch airmode motor stop does not have throttle above 1350 because I had airmode on a switch in that one, and I turn airmode on and off with the switch. Because motor stop is on in that one, you can tell I flip the switch to turn airmode on at roughly 12 seconds in. 

The other two have feature airmode on, so it is not on a switch, and so I increased the throttle to get it to go above the airmode activation point.  I do not see any change in the flightmodeflags from arming, through revving the throttle up, and through disarming at the end in the logs that have airmode set as a feature (not as a switch). Am I misreading the logs?

borisbstyle, 2016-06-19:
@mike-levenick
Even with airmode on switch airmode really gets activated above throttle
threshold! It works exactly the same as the feature, but the one can be
disabled in air and the other one is always enabled ;)

mike-levenick, 2016-06-19:
@borisbstyle that's odd. If that's the case, why is it that the motors begin spinning at min throttle when I flip my airmode switch? Nothing else changes, but I flip my airmode switch, and the motors spin at min throttle (even with feature motor stop turned ON).

borisbstyle, 2016-06-22:
@mike-levenick 
Airmode overrules the motor stop. But I see that there is indeed a small dependancy from switch in some scenarios.

I will fix that in 2.8.2

isilcala, 2016-06-23:
I believe mike-levenick was saying, if now airmode really gets activated above throttle threshold even when it's on a switch, then the motors should not begin spinning at min throttle. Once the throttle go above the threshold then airmode overwrites the motor_stop. But that's not the case what mike-levenick and me have been able to see.

airmode on feature:

1. when first armed, throttle = min, motors should not be spinning. (True)
2. throttle goes above threshold, airmode should be activated. (??)
3. throttle goes back to min, motors should keep spinning as airmode is activate. (False, motors stop spinning)

airmode on switch, if as you said 

> Even with airmode on switch airmode really gets activated above throttle threshold! It works exactly the same as the feature

then it should be like this :

1. armed, switch to airmode, motors should not be spinning. (False, motors begin spinning)
2. throttle goes above threshold, airmode should be activated. (??)
3. throttle goes back to min, motors should keep spinning as airmode is activate. (True)

borisbstyle, 2016-06-23:
No thats dangerous. When you fly with airmode ON motor stop is always
overruled. Airmode ON doesnt mean it is active yet.

Imagine not knowing if airmode is activated or off and still disabling
motors in air and crashing.

borisbstyle, 2016-06-23:
so short summary.....when airmode switch is on motor stop is already
overruled.

What do yoh want to make your motors start at 1350 and shoot your quad in
the air :)

mike-levenick, 2016-06-23:
@borisbstyle - correct. 

Dead horse being beaten, I think Boris gets it.

I am using "are the motors spinning" as a diagnostic test, since the motors do (and I believe should) spin at min throttle when air mode is activated, even if motor stop is on.

When I use switch, it seems to work correctly. If motor-stop is on, and I activate airmode via switch, the motors begin to spin at min throttle, telling me air mode is active.

When I use feature air mode and motor stop is on, I throttle up past the activation point, and then throttle back down, and the motors stop--telling me air mode is not activated properly. 

Since the purpose of airmode is to give increased pid control at low/zero throttle, I assume that airmode is not meant to deactivate when you go below the activation threshold, as that would defeat the purpose, which tells me airmode is not activating properly when using feature.

So yes, to say that there are times it is dependent on switch sums up perfectly what I'm saying. Switch appears to be working correctly (except that if it's not supposed to activate until you go above 1350, even with switch, I believe it is activating immediately upon turning the switch on. My motors begin to spin at min throttle the second I turn on airmode switch). 

Thank you for your hard work Boris.

borisbstyle, 2016-06-23:
@mike-levenick 
Yes, but the feature thing is also being fixed in 2.8.2. Feature airmode will fully overrule motor stop

isilcala, 2016-06-23:
so the bug actually is motor_stop was not correctly overruling by airmode at min throttle, besides that the feature airmode was working well, am I understanding this right?

and as always, thank you so much for bringing us this great betaflight :)

mike-levenick, 2016-06-23:
The bug was that feature airmode was not correctly activating air mode under the circumstances that it should.

Motor stop was used by me purely as a diagnostic test, as the motors should not ever be stopped when air mode is active. If the motors are stopped, air mode is not correctly activated.

isilcala, 2016-06-23:
I see this commit:
https://github.com/betaflight/betaflight/commit/9160a2adbc1a6167632d74f280738a5e7eca9187

read the code we can see the bug is motor_stop was not correctly deactivating, but from your description and your blackbox log (thank you for that, my F1 board doesn't feature blackbox) the bug was airmode was not correctly activating. seems there is still a gap there.

mike-levenick, 2016-06-23:
To clarify, airmode should overrule motor stop.

If you have motor stop active, and sit at min throttle, the motors are stopped. However we don't ever want the motors to stop spinning with air mode on. So if feature motor stop is on, and air mode is active, the motors will spin at min throttle. 

His comment "fix motor stop not being deactivated with airmode feature" means that airmode is not correctly overruling motor stop. 

---

As an aside, I do believe that motor stop is not the only issue, and that feature airmode is not activating airmode at all for me, as my flips and rolls are significantly faster with airmode on a switch than they are with feature airmode (even with feature motorstop disabled). I was simply using motorstop to help diagnose the issue.

That said, I'm 100% confident in the fact that Boris knows what's up now, and I think he's on it.

isilcala, 2016-06-24:
yes I believe we are on the same page now. thank you both.

closed 2016-07-03 by borisbstyle

mike-levenick, 2016-09-12:
Hey @borisbstyle and others, not sure if this was touched at all during the betaflight 3.0 code cleanups, but as mentioned above, I believed the airmode in 2.8 bug was not fully resolved--and that feature airmode was NEVER activating (at least on calibri race).

I have a buddy with a TBS Vendetta running 2.9.1, and his OSD shows that airmode is never activating with Feature Airmode, and we noticed an appreciable increase in rotational speed at zero throttle with airmode on a switch.

I'm working on getting a DVR from him.

borisbstyle, 2016-09-12:
There is actually no bug in airmode. It always gets activated when you
activate it properly.

mike-levenick, 2016-09-12:
Is there a reason that the OSD would display air mode as on when it is on a switch, but not when it is activated with Feature Airmode? 

The OSD as well as my blackbox logs show no change in flight mode when using feature airmode, and my rotational rates seem much slower than they do when it is on a switch.

borisbstyle, 2016-09-12:
It sounds like an error in OSD.

borisbstyle, 2016-09-12:
Airmode gets active above airmode_throttle_threshold btw.

mike-levenick, 2016-09-18:
I'm sorry, I know this is a dead horse and all that, but is blackbox supposed to see that feature airmode activated airmode? Because it doesn't.

When I turn feature airmode off, and set it to a switch, blackbox reports it correctly. 

I will attach two screenshots of my Motolab Cyclone on Betaflight 3.0.0, the first is with feature airmode, the second is with airmode on a switch. Feature airmode is not activating airmode according to the blackbox logs. You can see that I certainly went above my feature airmode activation threshold as well. 

I'm more than happy to do any other testing, or send you logs or anything you need. 

OSD says feature airmode doesn't work, blackbox says feature airmode doesn't work, and my subjective opinion is that my PID controller does not have as much control at min throttle with feature airmode as it does with airmode on switch (slow wobbles on zero throttle falls, slower rolls, etc.).

**Feature Airmode:**
[image]

**Airmode on switch:**
[image]

mike-levenick, 2016-09-18:
Here's a video showing the log header information, as well as the flight mode flags for both logs: https://www.youtube.com/watch?v=AR-ghOrZOcI

borisbstyle, 2016-09-18:
When feature enabled airmode is always enabled. No need to show it anywhere.

Its not a flight mode flag anymore, but default mode. You can see it in the
info page of blackbox with all of the settings.