betaflight/betaflight #463: Default branch should be renamed to `master`

martinbudden, 2016-06-09:
It seems there has been a bit of a reorganisation. Let's take advantage of this to rename the default branch `master`. I've been caught out a few times by accidentally rebasing a PR to `master` rather than `betaflight` and I bet I'm not the only one...

borisbstyle, 2016-06-09:
yeah was thinking that too. Maybe we can do that.

blckmn, 2016-06-10:
given this is still a fork of cleanflight, might need to add a cleanflight_master branch to ensure separation (cherry picking etc)?

borisbstyle, 2016-06-10:
the divergion did became big for even simple cherry picks, but I indeed
kept the original cleanflight master to be easily able to work on both
projectd.

AndersHoglund, 2016-06-11:
This is a sensitive issue. Just a deafult "master" is a bit too generic. We need to separate "cleanflight/master" and "betaflight/master". Especially when having multiple remotes. I have 3 in my local repo; my own github repo (cloned from betaflight), cleanflight and betaflight. In my local repo I also have a "hydra_master" nicknamed branch because I was not sure what "master" (cloned from betaflight/master) refered to. I pull cleanflight/master to "hydra_master" and betaflight/betaflight to "betaflight". Not using betaflight/master at all... 

I think integration and release should continue to be made on the "betaflight" branch, as it is by definition a fork and branch of cleanflight (the real master). Two or more "masters" can be very confusing.

martinbudden, 2016-06-11:
@AndersHoglund , agreed that confusion can occur.

However master refers to the branch, not the fork. Generally every fork has a master branch which is treated differently by GitHub. And suppose hydra created a branch called "development" (which is certainly possible), then we'd have to deal with two branches named development in different forks. And although unlikely, there is nothing stopping hydra creating a branch called betaflight - it could be used for staging changes that he wanted to merge back from betaflight, for example.

The fact of the matter is that we have to deal with branches with the same name in different forks.

AndersHoglund, 2016-06-13:
Yepp, that is the way with distributed SCM....

I have added a Jenkins "BorisB_BetaFlight_Development" job. Seems this is the integration-branch where the action is now and where CI is really needed, The betaflight-branch is the release-branch. [http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/](url)

borisbstyle, 2016-07-03:
I will close this issue to prevent cluttering

closed 2016-07-03 by borisbstyle