betaflight/betaflight #470: CC3D betaflight has a mind of its own

purplehazezzz, 2016-06-10:
Hi, I just finished my backup built. first the setup:
Martian V2 220 frame
Litllebee 20A blheli (thought version14.0)
Cobra 2204 2300KV
CC3D betaflight 2.8
oneshot125 enabled
damped light enabled
Orange dsm sbus receiver
Luxfloat 

Oke first of all it fly's great, slow rolls and flips no problem. But when I flip fast or when I roll fast it goes out of control. 

Like when i whant to do a really fast half roll to inverted, the quad suddenly rolls 1,5 or even 2,5 rolls. The same with flips.

So it seems to have problems with really fast stick input.

Any ideas? Ik tryd different pid controllers but that didn't work.

Thank you

jeto, 2016-06-10:
What are your rates and expos? Did you notice super expo is on by default in 2.8.

KiteAnton, 2016-06-23:
@purplehazezzz could you provide more info regarding your issue/behavior? Video/blackbox? Did you see jetos comment and are you aware of the super expo feature?

KiteAnton, 2016-06-27:
Old issue and no response from author, closing.

closed 2016-06-27 by KiteAnton