betaflight/betaflight #472: Compressing on Autobuild

borisbstyle, 2016-06-10:
@AndersHoglund
Would it be possible to compress all files in a zip/tar file after autobuild?

AndersHoglund, 2016-06-10:
Sure, I'll look into that... Probably needs a specific build-step, or a new plugin. Now all artifacts are just archived automatically by Jenkins by a simple tick in a checkbox. Don't think there is a builtin compress option.

AndersHoglund, 2016-06-10:
Second thought.... Github releases are not distributed compressed. Maybe we should keep things the same in all (both) places, not confusing the users.

AndersHoglund, 2016-06-11:
Done. 
Next Jenkins build artifacts will be only downloadable as a "betaflight_<version>-<build-no>-<git-rev>.zip" -file. For example: "betaflight_2.8.0-212-ae2e483.zip", OK? 

I could publish all build artifacts as well, *.zip, *.bin and *.hex, not only the zip archive. Then people could choose, single binary or all of them zipped up.  
/A

borisbstyle, 2016-06-11:
> "betaflight_--.zip" -file. For example: "betaflight_2.8.0-212-ae2e483.zip",

Zip is fine I think. The list is getting too long....lol.

Thanks for fixing it man!

closed 2016-06-19 by AndersHoglund