betaflight/betaflight #476: Throttle issues...

RipperDrone, 2016-06-11:
Throttle attitude has somehow changed since 2.8.1:

1. Max throttle seems to not unleash full power anymore (DYS SN20A, DragonFly 1806 motors) - like stuck at 70% max

2. When swapping btw Level and Rate mode, there is an offset in throttle of ~30% between each other. Cannot be compensated by throttle trim, tried it both ways (adjusting the high throttle mode = rate dwonwards and vice versa). Is there any new algorithms in throttle offset / evaluation?

borisbstyle, 2016-06-11:
Nothing changed in throttle actually.

Must be something else. Do you have a blackbox?

RipperDrone, 2016-06-11:
Then I have to carefully review my motors - maybe there's a short / low resistance issue limiting the rpm_max for one motor, forcing the others to slow down as well...

For now, I'll close this issue report item...

closed 2016-06-11 by RipperDrone