betaflight/betaflight #486: All targets need flight testing

martinbudden, 2016-06-12:
There have been some quite pervasive changes in betaflight to do with adding F4 support and the IO driver updates. While these have been done with care, there is always the possibility of mistakes having been made. I don't imagine anyone making the changes has access to more than a handful of different flight controllers, so not all of the changes have been flight tested.

This issue is a placeholder for flight testing. If you can flight test one of the flight controllers can you do so, and then as a comment to this thread say which flight controller you have tested.

nathantsoi, 2016-06-12:
awesome work all! here's the list of targets to test. I'll try to test some f3 targets today.

(re-gen with: `find src/main/target/* -type d -exec sh -c 'echo - [ ] $(basename {})' \;`)

- [ ] ALIENFLIGHTF3
- [ ] ALIENFLIGHTF4
- [ ] BLUEJAYF4
- [ ] CC3D
- [ ] CHEBUZZF3
- [ ] CJMCU
- [ ] COLIBRI_RACE
- [ ] DOGE
- [ ] EUSTM32F103RC
- [ ] FURYF3
- [ ] FURYF4
- [ ] IRCFUSIONF3
- [ ] LUX_RACE
- [ ] MOTOLAB
- [ ] NAZE
- [ ] NAZE32PRO
- [ ] OLIMEXINO
- [ ] PORT103R
- [ ] REVO
- [ ] RMDO
- [ ] SINGULARITY
- [ ] SPARKY
- [ ] SPRACINGF3
- [ ] SPRACINGF3EVO
- [ ] SPRACINGF3MINI
- [ ] STM32F3DISCOVERY

mikeller, 2016-06-12:
@martinbudden: Happy to help with flight testing. Are these changes in the `betaflight` branch already, or should we build from the `development` branch? (Also, a script (or instructions) to build all targets that are currently meant to build on a branch would be appreciated.)

borisbstyle, 2016-06-12:
@martinbudden
Flight testing is the least of my worries.

It is the stuff with IOs.....serial comm and all other features what I
would like to see tested.

For example just before I went away for holiday I quickly tries dev branch
on my COLIBRI_RACE and the leds were not working, but everything seemed
fine.

Not sure if that was fixed later.

martinbudden, 2016-06-12:
@borisbstyle , when I said "flight testing" I sort of meant "go out and fly it and report any problems", but I guess I wasn't specific enough.

When I was cleaning up the makefiles and targets I noticed quite a few of the targets wouldn't build because not all the pins had been converted to the new IO system. I corrected these, so everything builds, but of course mistakes could have been made by me or anyone involved in the IO conversion. So I guess I am agreeing with you.

mikeller, 2016-06-13:
Just gave 3.0 a quick spin on my NAZE32 rev 6.

The IO seems to be mostly working, except for the beeper, which does not work at all.

Also, after about 30 seconds flight time, the motors suddenly stopped, and the craft dropped out of the sky. Unfortunately I did not record logs at the time, and now the craft needs repairs before I can do more testing...

borisbstyle, 2016-06-13:
Those are great test results.

I do recommend more bench testing before real flying on this one. Let it
fly without props etc. If it survives that it will work in air as well.

mikeller, 2016-06-13:
I **did** give it a quick spin with props off to verify all TX input and gyro stabilisation were working before I took it outside. But apparently the spin wasn't long enough, or the dropout was triggered by something like gyro input...

tannewt, 2016-06-13:
LED toggling wasn't working until 46941032 afaik. On my own FC i2c isn't working and passthrough isn't working. i2c work for anyone else?

mikeller, 2016-06-14:
Tested some more, and came across this as part of `dump all` on a vanilla 3.0 configuration:

```servo 0 -5912 -12285 -9209 100 90 5 90
servo 1 -5912 -12285 -9209 100 90 5 90
servo 2 -5912 -12285 -9209 100 90 5 90
servo 3 -5912 -12285 -9209 100 90 5 90
servo 4 -5912 -12285 -9209 100 90 5 90
servo 5 -5912 -12285 -9209 100 90 5 90
servo 6 -5912 -12285 -9209 100 90 5 90
servo 7 -5912 -12285 -9209 100 90 5 90
smix reverse 0 0 r
smix reverse 0 1 r
smix reverse 0 2 r
smix reverse 0 3 r
smix reverse 0 4 r
smix reverse 0 5 r
smix reverse 0 6 r
smix reverse 0 7 r
smix reverse 1 0 r
smix reverse 1 1 r
smix reverse 1 2 r
smix reverse 1 3 r
smix reverse 1 4 r
smix reverse 1 5 r
smix reverse 1 6 r
smix reverse 1 7 r
smix reverse 2 0 r
smix reverse 2 1 r
smix reverse 2 2 r
smix reverse 2 3 r
smix reverse 2 4 r
smix reverse 2 5 r
smix reverse 2 6 r
smix reverse 2 7 r
smix reverse 3 0 r
smix reverse 3 1 r
smix reverse 3 2 r
smix reverse 3 3 r
smix reverse 3 4 r
smix reverse 3 5 r
smix reverse 3 6 r
smix reverse 3 7 r
smix reverse 4 0 r
smix reverse 4 1 r
smix reverse 4 2 r
smix reverse 4 3 r
smix reverse 4 4 r
smix reverse 4 5 r
smix reverse 4 6 r
smix reverse 4 7 r
smix reverse 5 0 r
smix reverse 5 1 r
smix reverse 5 2 r
smix reverse 5 3 r
smix reverse 5 4 r
smix reverse 5 5 r
smix reverse 5 6 r
smix reverse 5 7 r
smix reverse 6 0 r
smix reverse 6 1 r
smix reverse 6 2 r
smix reverse 6 3 r
smix reverse 6 4 r
smix reverse 6 5 r
smix reverse 6 6 r
smix reverse 6 7 r
smix reverse 6 7 r
smix reverse 7 0 r
smix reverse 7 1 r
smix reverse 7 2 r
smix reverse 7 3 r
smix reverse 7 4 r
smix reverse 7 5 r
smix reverse 7 6 r
smix reverse 7 6 r
smix reverse 7 6 r
smix reverse 7 0 r
smix reverse 7 1 r
smix reverse 7 2 r
smix reverse 7 3 r
smix reverse 7 4 r
smix reverse 7 5 r
smix reverse 7 6 r
smix reverse 7 7 r
```

MJ666, 2016-06-14:
Have tested the BlueJayF4 target last days quite intensively and make a few flights nearly every day. i use an modified target with SDCard disabled and SPI flash enabled (BlueJayF4 Rev1). Here are the actual problems I have seen:

-    CLI dump is missing major parts. Single commands and small output of a few lines are ok (like tasks). Sometime ther I something still in the buffer an will only come after an new input.
-    Ports page in configurator is hanging.
-    BlackBox page in configurator is hanging (only since an very recent build).

VirtualEnder, 2016-06-15:
Just flashed and tested the Revo F4 clone and everything seems good! Baro and Mag were even working and recognized on the bench.  Serial works fine, ports works fine.  The only bug I see is that it doesn't reboot properly in the configurator, I just have to manually disconnect and reconnect though and it's fine. Also setting looptime from the GUI didn't seem to work, but when I set it from the CLI it worked fine.  Also when I tried it again from the GUI it seemed to work ok, so it may have been a hiccup.  So far everything seems to be working fine, to the point where I might flight test it later.  Great work guys!

mikeller, 2016-06-16:
Beeper for NAZE32 is now fixed in #523. But there seems to be a problem with initialisation on power-up: #525.

VirtualEnder, 2016-06-18:
Ok, I tested my Emax variant of the BlueJay and it has all kinds of issues.  

1. Output #4 on oneshot 25 outputs a signal about 35uS longer than #1-3.  When I switch to multishot it goes away and all outputs are fine.
2. The align_gyro and align_acc commands don't work.  They save properly and the show the correct values in the CLI, but they don't actually change the board orientation.  Setting board orientation does work however.
3. This is the really weird one.  The gyro sync and pid sync denoms are really having problems.  At 1:1 (8khz synced) MCU usage is about 100% and everything in the system starts lagging, serial gets buggy, etc..  Even after a while it's still at about 5khz from tasks. At 2:1 (4khz synced) it runs OK, and MCU usage is only around 8%, but when I first connect usage is 100% and looptime is stuck at 1000uS.  After I click around a bit it stabilizes at 250uS. 

Don't think I trust it enough to try flying it yet :D  This was on the latest Jenkins build #27 adb38ca.

RC-Slater, 2016-06-18:
does that looptime deviation when you first connect last for approximately 15 seconds? If so it could be the issue with the software constantly checking for SD card. Try with an SD card inserted and see if it still happens.

VirtualEnder, 2016-06-18:
yes, it's about 15 seconds, but this variant doesn't actually have an SD Card slot. :( Maybe later on I'll try to compile one with the SD card support disabled.

kc10kevin, 2016-06-18:
That is the SDCard settling issue.  With it defined in the target, it is searching for the SDCard.   Even if there is not an sdcard slot on the board.
It tried 8 times at 2 second intervals.  
This is something that affects all SDCard targets when no SDCard is present in the slot. 
Something that needs to be addressed quickly.  

Easy fix in the code is to change the delay between retries down to 200ms from 2000ms.  This makes the delay only 1.6 seconds which is not really noticeable.  I've tried it with the delay of 200ms and it works fine.  Even when a fresh card is inserted.

I'm sure there is a better way of doing this.  Like checking one time if a card is inserted and giving up after that.

VirtualEnder, 2016-06-18:
This doesn't explain the 100% MCU usage at 8khz/8khz though, that combined with the lack of rotation support makes me think there is still something going on with the IMU driver on this target.  I would be interested to see if it is just this Emax FC, or if the genuine BlueJay has the same issues.  Unfortunately I only have this one here. Anyone else tried the genuine BlueJay?

MJ666, 2016-06-18:
I'm flying an BueJayF4 Rev1 since a while with an custom build of nearly dayly firmware updates. I build the firmware with SPI flash support insted of the SDCard. The FC is build into the copter upside down and I dont have an issue with the related CLI commands. I use oneshot125 with Kiss ESC's and have not seen problems (I dont have measured the pulses). I have not played with the looptime yet since its flying nicely without changes. Details with the configuration can be found here. http://www.multiwii.com/forum/viewtopic.php?f=12&t=6456&p=67740#p67694
http://www.multiwii.com/forum/viewtopic.php?f=12&t=6456&p=67740#p67737

nathantsoi, 2016-06-18:
@tannewt how can you tell i2c is breaking and on what target? my spracingf3 seems ok. thanks for confirming passthrough isn't working, it isn't working for me either. i'm looking at that now

nathantsoi, 2016-06-18:
@4712, re passthrough programming. any advice?

[image]

tannewt, 2016-06-18:
@nathantsoi Its my own FC that will be released in a couple weeks hopefully. It uses i2c to read an EEPROM with configuration info and that wasn't working last I checked. I plan on circling back to it in a couple days after I wrap up test jig stuff. It could be my code so don't worry about it now. Knowing it works for you means its likely on my end. I had to add another i2c read method for longer addresses.

I'll also circle back on the passthrough stuff if you haven't solved it by then. :-) I have a week-ish to finalize the firmwares to ship the boards with so it'll be a scramble to get Betaflight 3 ok enough to ship on both my F3 board and F4 board. The packaging and instructions will say to update it but you can't rely on people following instructions. I'm actually tempted to prevent arming all together to force people to update...

nathantsoi, 2016-06-18:
ok, esc passthrough programming on f3 is fine, my minimosd was plugged into the msp port and powers on when the battery is connected :)

my f3 test quad wobbles like crazy i cant even get it off the ground. it may be a tuning issue, but looking at the blackbox, it seems like angle mode is enabled, even though i have not flipped the angle mode switch. does anyone else see this? i'll dig into the code in a bit...

[image]

MJ666, 2016-06-21:
Here is some update on my recent testing’s:

BlueJayF4 Rev1: Lots of flight testing during last days. Using KISS ESC’s (12A V1.2, LPF disabled via solder bridge) with Oneshot125 and an LemonRX sat. No issues during 30-40 batteries over the last 1.5 weeks. The only known issue is the non-working CLI Dump (partially missing output). This problem looks to be only affecting F4 targets. Looks also similar what @mikeller has reported.

AlienFlight F4 brushed: Flight tested. Looks to be flying ok but not really tuned yet.

AlienFlight F4 brushless: Oneshot125 output looks not really working properly. Can even not calibrate ESC’s (KISS 12A V1.1, bridge soldered to Oneshot). 2 ESC behave completely different than the others. May need to check output with the scope. Using same sating like on the BlueJayF4:

    motor_pwm_protocol = ONESHOT125
    motor_pwm_rate = 2000

AlienFlight F3 V2: bench tested only. Looks to be ok

AlienFlight F3 V1: The I2C driver has a problem. MCU is not detected. The definition in the I2C driver file will take priority over the definition in the target file for unknown reason. If I change the default in the I2C driver the FC looks to be working (bench test only). Likely the same problem @tannewt reported with his new FC and I2C.

    #ifndef I2C2_SCL 
    //#define I2C2_SCL PF4
    #define I2C2_SCL PA9
    #endif

martinbudden, 2016-06-21:
From a quick glance it looks like `target.h` should define
```
#define I2C2_SCL        PA9
#define I2C2_SDA         PA10
```
rather than
```
#define I2C2_SCL_PIN         PA9
#define I2C2_SDA_PIN         PA10
```

Does that seem right. If so can you try it and report back?

borisbstyle, 2016-06-21:
@blckmn 
I am thinking to add back the unsynced_pwm option as this comes nice in the new configurator.

I still didnt have much time to test with 3.0, but will start this soon. I first want to polish the 2.8 stuff to add all these crazy cli options into configurator.

blckmn, 2016-06-21:
@borisbstyle no worries, it does make it clear I guess. When I fix up the merge I'll pop it back in.

MJ666, 2016-06-21:
@martinbudden This fixed the I2C issue of the AlienFlight F3 V1. Thanks, sometimes I'm blind. :)

blckmn, 2016-06-21:
@mj666 @martinbudden the definition issue with i2c dies highlight an inconsistency in naming. Some have _pin others don't. e.g SPI versus LED versus i2c might need a future cleanup

martinbudden, 2016-06-21:
I agree we should sort out this consistency in naming. Inconsistency in naming can result in silly, but hard to track down bugs.

tannewt, 2016-06-22:
My i2c issue on my F3 was also the _PIN suffix plus the change from USART1_TX_PIN to UART1_TX_PIN which by default was my i2c data line. :-(

blckmn, 2016-06-22:
Ok. I'll clean them up.  What's the verdict pin or no pin in the naming?

tannewt, 2016-06-22:
I was gonna offer to clean them up too after we decide. I vote no pin because longer term we should just define pins and not periphs and clocks. So, there will be nothing to differentiate against.

martinbudden, 2016-06-22:
I'm not sure about pin or no pin. I started cleaning it up with no pin myself, and the code didn't look quite right. After all PB4 (say) is a pin, and it encapsulates all the info about that pin.

So, @blckmn , can you hold off a little while and let me think about this a bit more. Doing the cleanup is a bit of a tedious activity, so it's worth spending a bit of time thinking about it first.

One thing that could be cleaned up is the use of `pin` in `timerHardware_t`. Currently we have:
```
typedef struct {
    TIM_TypeDef *tim;
    ioTag_t pin;
...
```
can you change that to:
```
typedef struct timerHardware_s {
    TIM_TypeDef *tim;
    ioTag_t tag;
...
```
That is two changes:
1. add the `timerHardware_t` label
2. change the `pin` item to `tag`
This will avoid rather confusing expressions such as:
```
IO_PINBYTAG(timerHardwarePtr->pin)```
```
which would then become
```
IO_PINBYTAG(timerHardwarePtr->tag)
```
which is just much better to read. Eventually these will all go, but until they do, having this change will reduce the number of errors made in the conversion.

mikeller, 2016-06-22:
Fixed my tricopter tonight, put 3.0.0 onto it, and gave it a quick spin on the bench and then outside. The yaw PIDs seem to be off compared to 2.8.0 that I was flying before the tail servo burnt out, but apart from that it was flying fine.

One thing that I noticed was that the servo mixer configuration initially dumped as those crazy values that I'd reported earlier on. This seems only to be happening right after flashing. Once I'd disconnected power, then reconnected, the servo mixer configuration was fine.

blckmn, 2016-06-22:
@martinbudden Good feedback. I'll make that change a pull request tonight.

blckmn, 2016-06-22:
@martinbudden have added that change to the vcp_fixes pull request. I am with you as it needs more thought. Although in the target.h file having a PA1 describes the pin and is obvious.

BUT

Elsewhere reading the code it is hard to see that I2C_SCL or LED0 are a pin definition (PA1 becomes hidden).

So I am on the fence.

martinbudden, 2016-06-22:
That's the thing, in the `target.c` file it is obvious. But in the driver code it can get confusing.

tannewt, 2016-06-22:
Anyone get i2c working on an F4? I'm struggling to get it working and I'm making it worse by trying to change it to support two byte subaddresses for reading EEPROM. Looks like we're using the F1 code but I'm not sure its that simple since its doing register checks and such.

MJ666, 2016-06-22:
On my F4 dev board (AlienFlight F4 test board) I have an baro connected to I2C wich is working. Also my BlueJayF4 has an working baro connected to I2C.

tannewt, 2016-06-22:
Hrm. Must be my code then. I tried changing subaddress to two bytes but the
event interrupt handler is complex.

tannewt, 2016-06-23:
@MJ666 Which baros are they? I'd like to look at their I2C calls.

It seems like its a multibyte read thats causing me issues. Heres the end of the signal where it should continue to read but it appears the master doesn't ACK the slave. The previous packet set up the read with two bytes like I need and single byte reads worked earlier. The EEPROM isn't the problem because it works fine connected to an F3FC.

[image]

tannewt, 2016-06-23:
So, I figured one issue out. The I2C timeout is really short! The F3 code uses I2C_LONG_TIMEOUT at every stage of the I2C transaction while the F1/F4 code uses I2C_DEFAULT_TIMEOUT (aka I2C_SHORT_TIMEOUT) for the whole transaction! I'm hitting this issue because:
1. I'm doing longer reads and the timeout doesn't scale.
2. The F4s run faster than the F1s so the busy loop takes less time.

Lastly, I'm doing multiple I2C calls back to back so the timed out transaction thats still currently running gets clobbered by the start of the next transaction. The next transaction doesn't work, however, because there is static state variables in the interrupt handler that don't get reset by the previous transaction finishing.

The easiest fix is to raise the timeout but the better fix is to stop using interrupt driven I2C for F1 and F4. It doesn't buy us anything because our main code is just sitting in a busy loop anyway and its way more complicated. I'll follow up later today with a fix after I debug another failure. Thanks!

skaman82, 2016-06-26:
Just tried flashing BF 3.0 to my SPracing F3 boards. The Acro version work fine, but the Deluxe version doesn't boot at all. Only shorting the boot pins and reflashing with 2.9 helps.

borisbstyle, 2016-06-26:
I had something simillar with colibri race. But lux race target worked
there. I guess there is some feature broken used on those targets
specifically

MJ666, 2016-06-26:
@tannewt at the AF F4 I have an BM280. here looks to be also a problem now it is detected but I2C errors are counting and the sensor reading looks to be not correct. the BJ has an MS5611 and this is working.

tannewt, 2016-06-26:
@MJ666 The BMP280 does longer i2c read calls. My guess is that they
timeout. Try increasing I2C_DEFAULT_TIMEOUT.
https://github.com/chickadee-tech/betaflight/blob/ckd_targets/src/main/drivers/bus_i2c.h#L22

I also have a debug statement that seemed to magically fix things too:
https://github.com/chickadee-tech/betaflight/blob/ckd_targets/src/main/drivers/bus_i2c_stm32f10x.c#L199

MJ666, 2016-06-26:
@tannewt i will give this an try when im back. I will be traveling for busines for the next week and can not do any testing.

tannewt, 2016-06-26:
I'm in the same boat for the next ten days or so.

llambkin, 2016-06-28:
I have been flying my KISS FC for a few days now (probably about 10-15 packs) and its flying great! I do have to say it is good to be back :D I am sure that a LOT of people are getting frustrated with the KISS code, and will move back to Betaflight once 3.0.0 is stable.

I have added some stuff to be fixed (Vbat, motor order etc.) by @martinbudden on that thread. I dont know where the pinouts for things like LED strip would come from, but willing to test if someone can give direction, havent tested buzzer, I also cannot arm when I have selected 4khz. I dont know if this is a KISS FC thing or not.

martinbudden, 2016-06-28:
@llambkin , there is now a feature in betaflight that prevents arming if the CPU is overloaded. This might be what is happening when you set your KISS FC to 4khz. Use the `status` and `tasks` commands int the CLI to see if this is the case.

AndersHoglund, 2016-06-29:
CC3D does not boot with 3.0.0. Should it be reported here or an issue of it's own maybe?
All I get is 5 short and 2 long blinks, FAILURE_MISSING_ACC. 
Using CC3D.hex, (non OPBL), and flashing via main port. Last test on development branch,  Jenkins build 68 (2f10c77).

martinbudden, 2016-06-29:
@AndersHoglund , I think this should be reported as an issue on its own.

AndersHoglund, 2016-06-29:
OK, CC3D issue #636

Rossbow, 2016-06-30:
Just testing a Fury F4 on betaflight_3.0.0-dev-69-94f95d6
I can not seem to get telemetry working correctly my Taranis is reading 7.6v
Is this an issue or do I need to look at my hardware ?

blckmn, 2016-06-30:
Hi @rossbow that looks like it's the battery in the taranis itself. If it's coming via telemetry it would say VFAS... Have you run the discover new sensors yet?

Rossbow, 2016-06-30:
Hi@blckmn  yes you are correct about the battery in the taranis but there is also no vfas voltage shown beside it, the firmware in my taranis does not have discover new sensors I just enable fas in the telemetry menu and it normally works, well it does on my 5 other quads with F3 boards.
Not sure why this won't work

borisbstyle, 2016-06-30:
I hope to get some test results this evening. Been crazy busy with other stuff lately :D

carebear482, 2016-06-30:
Revo nano wont boot up with betaflight

Rossbow, 2016-06-30:
Thanks  @borisbstyle 
I flew a few batteries on the Fury F4 yesterday and it flys close to perfect on the default P's I tried raising them a little but it was not required, only needed to raise I term a few points everything elese seems to work fine except for telemetry.
Thank you all for you effort, it is appreciated

borisbstyle, 2016-07-01:
Ok guys finally took some time testing things yesterday on different targets. I have to say that I totally love the new IO structure. So much cleaner, easier to review target differences and adding new targets. Huge step IMO. I want to take everyone who contributed and especially @martinbudden and @blckmn.

I havent been able to test CC3D yet. Hope someone else can dig into it. CC3D has always been special.
 with many hacks.

On the bench all seems fine for now. I did see something weirs on vbat on my development custom target I used, but maybe thats something else I messed up. Need to dig into it.

I do see pointer errors during compiling for targets with softserial enabled.  That needs to be addressed, but also softserial functionality needs to be tested.

Next to do is flight testing, which I think it will be fine, but always needs to be done.

[36 more replies]

closed 2016-07-27 by borisbstyle