betaflight/betaflight #487: 2.8.1. Rateprofile dump/restore still not right...

RipperDrone, 2016-06-12:
Checked rateprofile in 'dump all' CLI listing. It does still not show the active rateprofile (e.g. rateprofile 0) as last command, instead it says:

# restore original rateprofile selection
rateprofile 2

Somehow it does not write the current rateprofile into the dump...
Using RC1.

nathantsoi, 2016-06-12:
btw, i'm working on a new configuration tool & command line thingy that will remove the need for the cli :)

it will be a while, since i need to re-emplement a bunch of stuff, but the cli will turn into a bunch of client-side terminal commands that operate off msp instead of the text cli thing.

lmk if anyone knows angularjs and wants to help

[image]

RipperDrone, 2016-06-12:
sry, don't know angular... (what???). It's ok to work around the bug so long...

mikeller, 2016-06-12:
Hi @RipperDrone.

This is weird, I've tested this on my HW and it works fine. One thing you need to keep in mind is that the rateprofile setting is stored **per profile**, so there is two `# restore original rateprofile selection` in the output of `dump all`:

```
[dump general config]

[set profile 0]
[dump profile 0 config]
[rateprofile 0 (profile 0)]
[dump rateprofile 0 (profile 0)]
[set rateprofile 1 (profile 0)]
[dump rateprofile 1 (profile 0)]
[set rateprofile 2 (profile 0)]
[dump rateprofile 2 (profile 0)]

# restore original rateprofile selection
[restore rateprofile setting for profile 0]

[set profile 1]
[dump profile 1 config]
[rateprofile 0 (profile 1)]
[dump rateprofile 0 (rofile 1)]
[set rateprofile 1 (profile 1)]
[dump rateprofile 1 (profile 1)]
[set rateprofile 2 (profile 1)]
[dump rateprofile 2 (profile 1)]

# restore original rateprofile selection
[restore rateprofile setting for profile 1]

# restore original profile selection
[restore profile setting]
```

i.e. the last `# restore original rateprofile selection` you see in the dump is not necessarily the rateprofile that will be active in the profile that is active after all of the dump has been restored.

If this doesn't solve it for you, please post a complete dump, plus the information what profile / rateprofile was active at the time you executed the dump, plus your hardware information, and I will look into this.

RipperDrone, 2016-06-12:
I see. However, after dump all, reload dumped data sequence into CLI command line, the last command is rateprofile 2, therefore when I enter 'rateprofile' CLI says 'rateprofile 2'. 

My profiles before dumping were profile 0 & rateprofile 0

mikeller, 2016-06-12:
@RipperDrone:

Ok, looks like the end of the `dump all` output gets cut off for you then, there should be more lines after the `rateprofile 2`:

```
[...]
# restore original rateprofile selection
rateprofile 2

# restore original profile selection
profile 0

# save configuration
save
```

(Explanation for the `rateprofile 2`: This is the rateprofile that is set for your profile 1.)

What FC hardware are you using?

RipperDrone, 2016-06-12:
SP3Racing

RipperDrone, 2016-06-12:
I double-checked the suspect behavior step by step:

1. In CLI, FC response to command 'profile' is: 'profile 0'
2. response to 'rateprofile' is: 'rateprofile 0'

So everything looks alright, FC is set to profile 0 & rateprofile 0..

Then continuing in CLI with:

'dump all' --> dumps all data, data ends with following lines:

'**# restore original rateprofile selection
rateprofile 2

# restore original profile selection
profile 0
# save configuration
save
#**'

As you can see, rateprofile 2 is restored without even re-loading the dump - it just happens in the process of dumping itself!

Consequently, at this point CLI responses are:

1. Command 'profile' --> CLI yields 'profile 0'
2. Command 'rateprofile' --> CLI yields 'rateprofile **2**'

Can you verify this sequence in your setup now?

closed 2016-06-12 by RipperDrone

mikeller, 2016-06-12:
@RipperDrone: Yes there is an issue with the the FC losing some lines when a dump is pasted into the CLI for restore. This is a problem that can only be solved in the cleanflight configurator, and there is a fix, but I am not sure if this fix is in the current release yet.
In the mean time, to avoid this the workaround is to split the dump into blocks, with each block having `profile X` as the last line, and then copy pasting it into the CLI block by block.

RipperDrone, 2016-06-12:
Great, will do ;-) Thank you very much.