betaflight/betaflight #498: Betaflight 2.8.1

marcodcribeiro, 2016-06-13:
Hi,

I put this version in my SPRacing, and begining have deadroll.
With the 2.8.0 i dont have deadroll.

Motor: SE2205
ESC: XM30a
FC: SPRacing

borisbstyle, 2016-06-13:
set anti_desync_power_step = 5000

marcodcribeiro, 2016-06-13:
Hi Boris,
I Seed that the 2.8.1 came with the anti_desync_power_step = 0, and i try put anti_desync_power_step = 10000 and have the same problem

borisbstyle, 2016-06-13:
Than your esc is faulty if it totally desyncs

marcodcribeiro, 2016-06-13:
any solution in the ESC? Recalibrate again? Flash again?

borisbstyle, 2016-06-13:
which one is it?

- Increase min throttle in betaflight a bit
- Lower startup power in blhelisuite of the esc if it is possible
- blacbox it

marcodcribeiro, 2016-06-13:
I Will Try.
I have Multishot and will try with oneshot125 too

tks.

marcodcribeiro, 2016-06-14:
Hi Boris,
I have try with OneShot (new oficial Blheli) but same problem.
So i Increase Min throttle in Beta, lower from 0.75 to 0.38 in blheli and set ppm min and ppm max in blheli same value to the four ESC (when calibrate not all ESC have the same values).
with this changes and 3 batteries i dont have the problem.
I will try more batteries tomorrow and give feedback.

My blackbox and video:  https://meocloud.pt/link/c082dff3-f9dc-44ec-9945-fc6ff4a31fa6/Boris/

marcodcribeiro, 2016-06-15:
Hi,

So make this changes:
Blheli:
 - Startup Power: 0.38
 - Motor Timing: Med/High
 - Demag compensation: High
 - PPM Min Throttle : 1020
 - PPM Max Throttle: 1980

Betaflight:
 - anti_desync_power_step = 1000 
 - Min Throttle: 1060
 - Max Throttle: 1940

5 Lipo without any deadroll. more lipos tomorrow to see.

trojanGoat, 2016-06-15:
>Did you say that your esc "don't" have the same values in blheli?  and this is what you have right now?

marcodcribeiro, 2016-06-15:
I fix the min and max in Blheli, but when calibrate in clean and go to blheli again some esc dont have the same min and max, for exemple ESC 1 (min: 1020, max: 1980), ESC 2 (min: 1024, max: 1992)

trojanGoat, 2016-06-15:
K.. I already tried doing that.  I haven't lowered startup power, but I don't see how this affects flight performance.  Isn't it meant for large props?.  I will try it one more time.  Did you end up sticking to Oneshot or Multishot?

marcodcribeiro, 2016-06-15:
I end up with Oneshot.
But now i try one motor at time in cleanflight with HOME and END key, and sometimes the ESC reboot.

So i go back to 14.4 testing and thats ok for now.

trojanGoat, 2016-06-16:
I also went back to 14.4 and no more desync problems.

marcodcribeiro, 2016-06-16:
Today i have try other change and now no problem with 14.6 and beta 2.8.1.
So e do this:
 - blheli 14.6 and make the calibration with blheli tab.
 - go to cleanflight and test one motor at time with HOME and END key.
 - see what motor do this: https://www.youtube.com/watch?v=PwhsReiCnHk.
 - in this motor or motors go to blheli and see if they have motor direction inverted.
 - if so put to normal and change the solder.
try.

For me works very well.

Issue closed

closed 2016-06-16 by marcodcribeiro