betaflight/betaflight #513: desync problem maybe ? idk

Mattawonga, 2016-06-14:
after alot of testing, i found out a weird problem on my new build.
dys xms30a and se2205/2300kv , tried betaflight 2.6.1 and 2.7.1
blheli 14.4, 14.5 , 14.6 and multishot 14.5, 14.6
at first i tought it was my Lux fc creating this trouble, but after swaping in a new Dodo rev3 in hope of better result, same problem is still there!!
when i try to fly more aggresive with fast stick reaction, my quad sometime loose control and crash always back right side.
when this happend flying higher altitude i can get control back by going zero throttle rapidly and then start flying again, but when too low i dont have time for this maneuver
i made a few videos flying Los showing the problem.
Would it be a bad esc? or a bad motor? 
will try swapping motor tomorow and see if the crash still happen on same side or not

tried lower min th and higher min throttle... lower max th higher max th,,, 

tried set anti_desync_power_step = from 100 to 3000... still same issue

you can see me doing some throttle jump and the last one does the problem and crash....
[blackbox_log_2016-06-14_171754.TXT](https://github.com/betaflight/betaflight/files/314991/blackbox_log_2016-06-14_171754.TXT)

Mattawonga, 2016-06-14:
more blackbox information

[blackbox_log_2016-06-14_173348.TXT](https://github.com/betaflight/betaflight/files/315006/blackbox_log_2016-06-14_173348.TXT)

marcodcribeiro, 2016-06-14:
Hi,

I have same setup (ESC and Motor) and same problem.
I change some settings (Boris input) and last lipo no problem.

So, do this:

trojanGoat, 2016-06-14:
I'm no blackbox expert, but it looks like overall your Pitch and Roll values are oscillating a lot before the crash happens. 

http://imgur.com/wTEDaSc

I have the same issue as you and I messaged Joshua Bardwell

He says... If It's always the same motor spinning up, it's likely a hardware issue with the esc or motor.  In my case it changes which means it's likely to be some sort of timing, demag or PID issue.

The betaflight anti_demag doesn't work for me either

Joshua says to go in the Cleanflight motor tab and bang the sliders up and down one at a time rapidly and see if the motor stalls and starts up again to reproduce the problem.

If the problem is reproduced, then the issue has to do with motor timing / demag
If the problem cannot be reproduced, the issue is PID related

Turn off heat protection on the esc ( <--------I'm not responsible for damage this might cause) make sure your min throttle is high enough.  

I will be trying out these fixes myself. Good luck.

100xanoff, 2016-06-19:
It seems that this is a problem with this - https://github.com/bitdump/BLHeli/issues/159
One variant - do not use a software reverse engines in BLHeli, change wires

100xanoff, 2016-06-19:
I helped replacing wires

Wesstone, 2016-06-19:
Im having a similar issue with the Aikon sefm Blheli_s escs.  Ive been troubleshooting intermittent death rolls for three days now.
Nothing is fully fixing this issue.  Ive tried everything I seen suggested on three different forums and here and it will temporarily correct the issue but comes back either in a different form or as the exact same issue.  
currently i have 
rs2205 2600kv
Aikon SEFM 30a 16.1
Dodo rev3a
BF 2.8.1 R2
min throttle 15 points higher than idle speed
anti desync power step at 10000
P gains are literally half default
startup power at the lowest setting
demag low
timing medium high

All of this stopped death rolls after a flip or roll but now its happening with a hard yaw input and I still have a ton of oscillations.

Is this a Aikon/blheli_s issue or BF or both?

Ive taken this FC off and put it another quad with littlebee 30a escs and dont have any issues besides the tune needing to be changed.

marcodcribeiro, 2016-06-19:
change te reversed to wire insted blheli.
works for me

DUNcANPELTENBUrG, 2016-06-19:
Same here! Problem solved by changing motor direction in BLHeli to normal on all 4 DYS XMS30A's and changing motor wires to change motor direction.

KiteAnton, 2016-06-19:
This seems unrelated to betaflight fw so I close this issue. 
If it is not reported to blheli already you should add an issue there instead.

closed 2016-06-19 by KiteAnton

Wesstone, 2016-06-19:
Except... I rolled back to 2.7.1 and have NO issues with no "workarounds" necessary.  Flew four packs.  Motors still reversed via software.  Reload 2.8.1 r2 and issues return.

borisbstyle, 2016-06-19:
@Wesstone 
Try 2.8.1 final

DUNcANPELTENBUrG, 2016-06-20:
@Wesstone In this case it's a DYS XMS30A / BLHeli issue. NOT a BetaFlight issue. Changing things in BLHeli or on the esc itself is not a workaround. The issue is addressed in the BLHeli WiKi as well.

husyildiz, 2017-04-22:
I was having same problem at mid throttle . After trying lots of solutions I finally discover power dist. is making magnetic or something like that interferes.