betaflight/betaflight #522: 2.8.0/2.8.1 motor control issue, noise/vibration confirmed

RipperDrone, 2016-06-15:
I have 2 SP3Racing ZMR250's now, issue with noise/low frequency vibrations confirmed on both.

This surely confirms earlier observation that from 2.8.0 onwards, thrust seems to show bad performance

- strange low-frequency humming / noise when throttling up >20%
- considerably less thrust, needs a lot (30%) higher throttle setting for hovering
- exists with default and intermediate (gyro_lowpass = 80 / dterm_lowpass = 100) noise settings 

Flashing back to 2.7.1 perfectly cures the problem.

Motor timing set to medium_high, damped_light in BLHeli. Have both 1806 Dragons and EMAX 2205 red bottoms with different ESC's, so it does not seem to be related to ESCs or motors specifically.

Anybody else having similar issues or any idea what this is about?

borisbstyle, 2016-06-15:
The vibrations were always there but before 2.8 the pids were lowered at
low throttle and now not anymore ;)

RipperDrone, 2016-06-15:
@borisbstyle 
Humming gets more pronounced @ higher / max throttle even. Not a particular low throttle issue. Quad is NOT instable to fly either. Just lacking thrust/power and humming like a bumblebee ;-)

If you think it's PID's, what's your recommendation how to compensate for it, just to give it a try and verify? I tried changing ALT/VEL PIDs from very low thru very high - no improvement... other PIDs I have not played around with too much yet...

borisbstyle, 2016-06-15:
use the same filter settings as in 2.7.1

gyro_lowpass and dterm_lowpass and try again

RipperDrone, 2016-06-15:
Just tried it, no difference. Nothing to do with gyro_lowpass / dterm_lowpass, obviously...

RipperDrone, 2016-06-18:
found 2 more pilots experiencing the same issue - here is one: 

https://www.facebook.com/groups/betaflight/permalink/292984484384645/?comment_id=293049527711474&reply_comment_id=293335174349576&notif_t=group_comment_reply&notif_id=1466200137568320

martinbudden, 2016-06-23:
@borisbstyle , I think this may be to do with the optimisation of the call to `updateRcCommands` that happened a while ago. Cleanflight had a similar problem caused by a similar optimisation.

If this is true then PR https://github.com/betaflight/betaflight/pull/566 should fix the problem.

@RipperDrone , could you build and test this?

RipperDrone, 2016-06-23:
@martinbudden , sounds like you spotted a very likely fix for the Baro issue. I'll be happy to do the test/verification but am still lacking some skills how to do the build.

I've done a decent amount of C++ coding long time back, but am not familiar with this environment here. Where can I find some instructions how to build specific interim versions from Pull Requests? Pls give me some guidance - I've only used BorisB releases & Jenkins Nightly Builds so far...

martinbudden, 2016-06-23:
If you known how to use git the process is quite easy (once you have set up the build environment). Clone the betaflight repository, merge in my changes and then type `make SPRACINGF3` on the command line.

If you don't know how to use git, than I think the simplest thing to do would be to clone the betaflight repository (just use the clone button in github) and hand edit in my changes - they are only a few lines.

Instructions on how to set up the build are at: https://github.com/cleanflight/cleanflight/tree/master/docs/development

I'm afraid that's all a bit terse, but hopefully should be enough to get you going.

RipperDrone, 2016-06-23:
Should do, will try. Thanks a lot. Not experienced with Git, will try the 2nd route…

RipperDrone, 2016-06-23:
Hi Martin,

 

spent another hour trying to set up the environment, still having limited understanding. Current setup is GitGui, GitBash and a nice GitHub Desktop interface. 

 

Have cloned-merged your commit #566 and actual betaflight onto my local Windows 10 computer. Still can’t figure out how to build from there. All command line attempts in Git Shell or Git CMD do not offer option to build. 

 

Am I missing out something evident here?

 

Pls do me a favor and build the SPRACINGF3 in your environment and send it to me then I can check / verify the BARO functionality immediately.

 

Thanks and regards,

Reinhard

martinbudden, 2016-06-23:
@RipperDrone , I've attached the file here:

[betaflight_3.0.0_SPRACINGF3.hex.zip](https://github.com/betaflight/betaflight/files/330813/betaflight_3.0.0_SPRACINGF3.hex.zip)

RipperDrone, 2016-06-23:
@MartinBudden,

 

thank you, I flashed your file onto my SPRACINGF3, since then I can’t get a USB connection made anymore, nor can I re-flash. Do I need a specific BetaFlight Configurator version? Tried with the latest Betaflight Configurator, tried 20+ times. Even downgrading to BF2.8.1 does not work anymore.

 

Any idea what’s going on now? USB connect has been flawless so far.

 

Have to set some parameters before I can do the BARO fix testing…

martinbudden, 2016-06-23:
I'm not sure what the problem is. Have you flashed other 3.0 development versions, or is this the first one you've tried?

RipperDrone, 2016-06-23:
Seems like weird things going on – after 50+ attempts including reset jumper, different combinations of Flash on Connect, No reboot sequence etc., I finally managed to flash the FC back to 2.8.1. Treid to up to 3.0.0 again, same result: No USB Connect anymore, no re-flashing without setting jumper and playing some 20+ times with plugging in/out USB and different port settings.

 

Seems like 3.0.0 with the BARO Fix needs other USB serial communication / other BetaFlight GUI version?

 

Sorry, cannot test BARO fix like that ;-(

RipperDrone, 2016-06-23:
1st ever.

martinbudden, 2016-06-23:
@RipperDrone , well thanks for trying anyway. It looks like there are some USB problems that need fixing....

RipperDrone, 2016-06-23:
you're welcome - yes seems to do with timing for serial USB connection. Whenever you have a BARO-fixed version available pls let me know when I can help test it. In parallel, I will try to see if I can find out how to set up a Win 10 build environment under GitHub (are you using CygWin?) to introduce mods into local builds for testing in the future...

martinbudden, 2016-06-24:
Yes, it's certainly worth you learning how to set up the build environment. I'm using a Mac myself.

mikeller, 2016-06-24:
@RipperDrone: Don't know if you still need it, but the next person having similar problems might find it useful: I've built a docker container that should make building Betaflight easy on any platform that can run docker: https://hub.docker.com/r/mikeller/betaflight-build/

RipperDrone, 2016-06-24:
Thank you. Tried it for some hours now:

 

-       Installed docker on Win 10

-       Cloned betaflight to my user directory

-       Tried to invoke your script as per instructions

 

Result on Powershell console: make: *** No rule to make target 'clean'.  Stop.

 

Did I miss a step here?

mikeller, 2016-06-24:
Hi @RipperDrone.

> Result on Powershell console: make: *** No rule to make target 'clean'.  Stop.

The error message suggests that you were not in the Betaflight source directory (the one that you'd cloned from GitHub). Also, I noticed that there was an error in the documentation, referencing NodeMCU where it should have been Betaflight.

RipperDrone, 2016-06-25:
Thank you - slowly getting there: 

Sorry to bother you again, but clone is detected now (I had wrapped the user directory name in quotes while not having spaces in my username, therefore the docker command went into Nirwana). 

 

Now compilation is running, however some warnings occur saying

 

warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7-m/libm.a(lib_a-sf_lrint.o) uses 4-byte wchar_t yet the output is to use 2-byte wchar_t; use of wchar_t values across objects may fail

 

At the end of the process, I do find a single file 

 

betaflight_2.8.2_NAZE.hex

 

in the obj folder, but all the other targets (like for my SPRacingF3) are missing.

 

Can you decypher this warning / misbehavior to help me get to a successful build?

 

Regards,

Reinhard

RipperDrone, 2016-06-25:
The documentation on how to parametrize your script for a certain platform is not clear to me:

Do I have to enter ‘docker run -e "<parameter>=SPRACINGF3" –e‘

and then (indicated by the ‘…‘) concatenate the rest of the build command?

Thanks,

Reinhard

_____________________

Options

You can pass the following optional parameters to the Docker build like so docker run -e "<parameter>=value" -e ....

*	PLATFORM The platform to build for. Use ALL to build for all platforms. (default: NAZE)

mikeller, 2016-06-25:
Hi @RipperDrone.

That's good success indeed, looks like your build system is working now. The compiler warnings are normal. The documentation is indeed not very clear, I will look into improving it.

RipperDrone, 2016-06-25:
Hi @mikeller 

I just completed testing of your commit #566 on SPRACINGF3. You nailed it! No more humming / low throttle throughput when BARO is on. Looks like this commit should be merged into future builds!

RipperDrone, 2016-06-26:
@mikeller have you changed any content of the prebuilt docker package to compile betaflight?

I tried to compile the actual 3.0.0 dev today, but wasn't lucky:

docker run -e "PLATFORM=SPRACINGF3" --rm -it -v //c/Users/Reinhard/Documents/GitHub/betaflight:/opt/betaflight mikeller/betaflight-build

yields: 

docker: An error occurred trying to connect: Post http://%2F%2F.%2Fpipe%2Fdocker_engine/v1.24/containers/create: open //./pipe/docker_engine: Das System kann die angegebene Datei nicht finden..
See 'docker run --help'.

Did I make a mistake?

mikeller, 2016-06-26:
@RipperDrone: I haven't changed anything, but it looks like dockerhub, where the docker repo is hosted, is having problems of some sort. Even though I have set the repo to be public, it asks me for authentication before being allowed to pull updates from the repo.

I hope this is only transient...

RipperDrone, 2016-06-26:
@mikeller - for some reason it's back to live, working again. Don't really know what it was, but anyway thanks a lot!

RipperDrone, 2016-07-01:
@mikeller forgive my ignorance, but I keep getting the 'time skew detected' warning messages every time I'm using your script. How can I avoid this? (just a nuisance, compiled .hex files seem ok...)

closed 2016-07-02 by RipperDrone