betaflight/betaflight #525: FC doesn't initialise properly / doesn't arm, NAZE32, 3.0.0

mikeller, 2016-06-16:
With latest in 3.0.0, I noticed that about 50% of the time when a battery is connected, the FC does not seem to initialise properly. It looks normal, and reacts to mode switches and stick input (as displayed on LED), but it won't arm (switch arming). As a workaround, when this happens, using the sticks to 'change' the profile to the one that was already active the last time it had been powered on seems to fix this. After the profile change, the FC arms (and flies) fine.

nathantsoi, 2016-06-16:
does this ever happen on usb only? (wondering if i can bench test w/o a battery?)

mikeller, 2016-06-16:
I did not notice any problems when I was working on the quad and had it connected on USB. But then, when this happens, the FC seems to be working fine, except that it does not arm. So it could well have happened, I would not have noticed.

blckmn, 2016-06-17:
I have a rev5 (with SWD connected) and a rev6 - so will test this on the weekend.

martinbudden, 2016-07-05:
@mikeller , are you still getting this problem?

mikeller, 2016-07-05:
@martinbudden: Sorry, I forgot about this one. This was fixed a day or so after I had reported it.

closed 2016-07-05 by mikeller