betaflight/betaflight #529: Smart port issue X4R, SPracing EVO

astajic, 2016-06-16:
Hi after download 2.8.1, set up my port as usual and I had a issue reading the voltage on my taranis, double check everything, but still the same. Re start the taranis and the quad and get the voltage back but just for few sec, then nothing.
Re flash with the 2.7.1 and the voltage back on my taranis, no more problem.

KiteAnton, 2016-06-16:
@astajic could you provide more information. I have ran many batteries on 2.8.1 on a SPracingF3 (not evo) without problem:

- Can you reproduce the error
- What is your dump settings, 
- What is your CPU usage, other features enabled?

astajic, 2016-06-16:
I've got to fly this afternoon but I will try again tonight
Cheers

RipperDrone, 2016-06-16:
not an issue on my SP3Racing

padmanek, 2016-06-16:
@RipperDrone Do you have EVO? @astajic specifically says EVO.

astajic, 2016-06-16:
Yes I tried with sp racing no problem, I've got the issue only with evo

RipperDrone, 2016-06-16:
no sorry, got the SP3 Racing Std / non-EVO

supermotodr, 2016-06-17:
I have sp racing f3 and f3 evo f3 is fine but evo only works on default looptime 2khz if I go to 4k looptime no fc telemetry,really annoying not able to use faster looptime.also with evo cannot calibrate esc but can with f3.

KiteAnton, 2016-06-18:
Seems like your evo is using to much cpu @4KHz and then the scheduler discontinues the lowest prioritized tasks (e.g. Led strip and telemetry). 

Check the outcome from 'status' and 'tasks'. If you haven't already disable unused hardware like acc/mag/baro etc that might use up unnecessary CPU.

supermotodr, 2016-06-18:
sorted my s port problem,wrong pid denom in cli now getting telemetry right up to 8k looptime.
still cant calibrate esc though even after changing looptime back to 1k.

closed 2016-06-18 by KiteAnton