betaflight/betaflight #531: Is it possible to set looptime via CLI?

voroshkov, 2016-06-16:
It looks like setting looptime via CLI doesn't work.
I tried:
a) looptime 500
b) set looptime=500
none seems to work.

Is it a bug or a feature?
Is it possible to add a possibility to set looptime via CLI?

KiteAnton, 2016-06-16:
It is no longer possible to set looptime directly in CLI. Use the configurator GUI to change looptime.

mikeller, 2016-06-16:
@voroshkov: However, it is possible to control the loop time from the CLI, by setting `gyro_sync_denom`, see http://www.rcgroups.com/forums/showthread.php?t=2464844&page=1176 .

borisbstyle, 2016-06-16:
yes it is

gyro_sync_denom
and
pid_process_denom

Peksy, 2016-06-17:
ok and which code is for looptime 2500 ?

KiteAnton, 2016-06-17:
@Peksy it is no longer possible to set looptime > 1000 us. Only allowed settings are multiplier (1-8) of 125us. Check the [wiki](https://github.com/betaflight/betaflight/wiki/Betaflight-specific-CLI-commands) for deeper knowledge/info.

Peksy, 2016-06-17:
KiteAnton thanks for explanation.

closed 2016-06-17 by KiteAnton