betaflight/betaflight #535: Hardware Revisions - should a build specifically deal with a hardware revision

blckmn, 2016-06-17:
At the moment there is a code associated with detecting hardware revisions etc - which is needed in a lot of places.

Suggestion for discussion:
Now that we have a new target directory based setup, would it be suitable to use the same (i.e. alternate targets) for hardware revisions.

e.g. NAZE_REV4.mk, add the REV4 define to device flags (adjust target.h based on the define), and then strip out any hardware revision specific code.

@martinbudden @borisbstyle your thoughts?

blckmn, 2016-06-17:
https://github.com/betaflight/betaflight/pull/536 is an example :)

blckmn, 2016-06-17:
NOTE: we would still need to have the target specific configuration options, but any if hardware_revision == x could be removed and replaced with a #ifdef

blckmn, 2016-06-17:
An example would be main.c:

`#ifdef USE_SPI
#ifdef USE_SPI_DEVICE_1
    spiInit(SPIDEV_1);
#endif
#ifdef USE_SPI_DEVICE_2
    spiInit(SPIDEV_2);
#endif
#ifdef USE_SPI_DEVICE_3
#ifdef ALIENFLIGHTF3
    if (hardwareRevision == AFF3_REV_2) {
        spiInit(SPIDEV_3);
    }
#else
    spiInit(SPIDEV_3);
#endif
#endif
#endif`

Would simply become:
`#ifdef USE_SPI
#ifdef USE_SPI_DEVICE_1
    spiInit(SPIDEV_1);
#endif
#ifdef USE_SPI_DEVICE_2
    spiInit(SPIDEV_2);
#endif
#ifdef USE_SPI_DEVICE_3
    spiInit(SPIDEV_3);
#endif
#endif`

and then in the ALIENFLIGHTF3 target.h

`#ifndev REV1
#define USE_SPI_DEVICE_3
#endif
`

mikeller, 2016-06-17:
Wouldn't this be a step backwards in reality?

It would go from a system where one binary was enough to support multiple revisions of the same target, to a system where you'd have to build / test / distribute separate binaries for different revisions of the same target.

Or to ask the other way round: What would be the benefits of this?

MJ666, 2016-06-17:
I think for me the goal for the detection was to use an single firmware for the different revisions. The user don't need to know which revision he is using and can simply flash the firmware for his board. Would be simple to have individual targets per revision but an hassle für the user.

mikeller, 2016-06-17:
@MJ666: Totally agreed. That's why I am questioning if moving the detection to compile time is a good move.

blckmn, 2016-06-17:
Yeah I'm on the fence (hence discussion). There's benefits both ways.

mikeller, 2016-06-17:
@blckmn: What's the advantages you see to having revision specific binaries?

martinbudden, 2016-06-17:
@blckmn , I'm with @MJ666 and @mikeller , here. For the user it's much easier if they don't have to worry about which revision of the board they are using. For developers it's probably easier too, since there won't be questions like "I flashed my Naze board and now it doesn't work" or "How do I find out which revision my board is?"

blckmn, 2016-06-17:
Yes definitely it is much better for the user. So what are the thoughts on how we move this code out of main.c?

```
#ifdef ALIENFLIGHTF3
    if (hardwareRevision == AFF3_REV_2) {
        spiInit(SPIDEV_3);
    }
#else
    spiInit(SPIDEV_3);
#endif
```

Perhaps an initConfig_t structure (rather than individual ones) that if USER_HARDWARE_REVISION_DETECTION is turned on, calls into the targetInitConfig(&initConfig) and any adjustments are made there - before subsequently calling init for the device?

```

#ifdef USE_SPI3
	initConfig.spiConfig.useSpi3 = true;
#endif

... all the configuration, but no initialisation ...

#ifdef USE_HARDWARE_REVISION_DETECTION
	targetInitConfig(&initConfig);
#endif

...

... the series of initialisation routines ...

if (initConfig.spiConfig.useSpi3)
    spiInit(SPIDEV_3);
```

blckmn, 2016-06-17:
Or do we have a targetSpiConfig(&spiConfig) routine - (and routines for any other device group, e.g. ADC, I2C, LED etc).

martinbudden, 2016-06-17:
@blckmn , generally a data-driven approach is better than a code-driven approach. So I think we should avoid the introduction of target callbacks where possible. In `config.c` the set of possible target-specific changes is large, so the use of callbacks is appropriate. Here I think the set of target-specific changes is small, so I think we can solve the problem by careful analysis and judicious choice of generic `#define`s.

Or to put it another way: code should be spherical - that it it should minimise the ratio of surface area [interfaces] to volume [functionality]. Callbacks with a small, highly specific purpose are like spikes on the surface of a sphere - they greatly increase the surface area to volume ratio and are inherently fragile.

blckmn, 2016-06-17:
Hi @martinbudden in that case - given the call backs - should we add whether or not spi3 is enabled / disabled as an option in config.c (and as such everything in main.c is effectively driven by config?)

This then simply makes use of the call backs that would need to exist already, and converts main.c into data driven. A simple config reset forces the target detection, writes the config to flash, and on boot no longer needs to perform revision detection?

blckmn, 2016-06-17:
Given it is already 90% config driven anyway ...

martinbudden, 2016-06-17:
Yes, I think moving it into config is probably the way to go.

I am, however, a bit wary of doing to much work in this area just at the moment. Changes to init and config can accidentally break things, and I don't think either of us have the hardware or the time to test all the different boards.

We are already in the desirable position whereby new targets can be added by just adding a new target directory, and we have shown this with a handful of boards.

There is nothing that is essential to get into the 3.0 release, and so I think I would be in favour of deferring further improvements to 3.0.1. I know we are on a bit of a roll at the moment and it is tempting to carry on, but sometimes it is best to stop and consolidate and then move on later.

I'm also wary of getting too far ahead of Cleanflight. I'm pretty sure when @hydra sees these changes he'll have suggestions for improvements and changes he'll want to make before accepting them into Cleanflight.

blckmn, 2016-06-17:
ok - sounds good. We can turn focus to getting the various boards tested and working correctly and getting the v3.0 release out there.

hydra, 2016-06-17:
I agree that it is too much hassle for developers to provide and support each hardware revision with a new binary. And too much uncertainty for users too.  Not sure how much first 1st line support you guys do with new users but I'm sure that we're all busy enough already, I know I am.

Would suggest stabilising the betaflight codebase so the we can get things merged and with the OSD support in CF which the users and developers seems to be extremely keen on.

blckmn, 2016-06-17:
Cool. Thanks for the discussion.

closed 2016-06-17 by blckmn