betaflight/betaflight #539: development question: logging system load & 100% utilization on startup

nathantsoi, 2016-06-17:
my apologies if this already has been asked and answered.

working on the osd code, i've noticed that every time i reboot the fc, for a few seconds after bootup, the `averageSystemLoadPercent` goes into the 700's for a bit (which can now be seen from the OSD)

maybe this is a silly question but does anyone know why this is? should it be addressed? and could this be related to: https://github.com/betaflight/betaflight/issues/525? seems like it might be confusing that some boards need a minute to settle, then can be armed?

more generally, is there a better way (better than the cli) to monitor system load over time?

```
# status
System Uptime: 18 seconds, Voltage: 361 * 0.1V (8S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8975
Cycle Time: 706, I2C Errors: 1, config size: 1316

# status
System Uptime: 20 seconds, Voltage: 361 * 0.1V (8S battery - OK), CPU:18%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8975
Cycle Time: 544, I2C Errors: 1, config size: 1316
```

blckmn, 2016-06-17:
Is this a target with SD card? There is an issue with SD card taking a while to detect and settle.

nathantsoi, 2016-06-17:
ahhh, cool. yep, that's pry it. sd card is not currently present... but it has a slot.

closed 2016-06-17 by nathantsoi

martinbudden, 2016-06-17:
I don't know why this occurs, but it should probably be investigated. It may be unimportant, or it may be like Cliff Stoll's [$0.75 accounting error](https://en.wikipedia.org/wiki/The_Cuckoo%27s_Egg).

You can view debug values in the configurator - set `debug_mode` to use them. `CYCLETIME` looks like the value to use. If not, you can temporarily change the instrumentation in the code to suit your needs.

blckmn, 2016-06-17:
I suspect it's if a card is not present it goes through a numbers of attempts. I'm pretty sure there's a nasty blocking loop in there ...