betaflight/betaflight #543: rx range bug?

Jahnkeanater, 2016-06-18:
When the radio outputs a value slightly out of the 1000-2000 range the flight controller appears to go into failsafe? All the channels go to 1500 until all channels return to the 1000-2000 range. 

Spektrum radios only output 1050-1950 and reserve +/- 50 for sub trims. This can be adjusted using the rxrange command but achieving 1000-2000 is still difficult without overshooting. Spektrum radios even when calibrated tend to give you a little more throw by pushing the stick against its endpoint. Calibrating to easily achieve 1000-2000 can result in the board freezing up when you push the sitck against its endpoint.

There should be a little bit of a buffer where the FC holds the value at 2000 or 1000 if the channel goes slightly out of range. A channel value of 2003 should be rounded down to 2000.

Arakon, 2016-06-18:
rx_min_usec and rx_max_usec set the point at which the signal is considered invalid and trigger failsafe.
And those are by default >100 points above and below 2000/1000.

KiteAnton, 2016-06-18:
@Jahnkeanater what values is sent from the radio when this happens ? If it is outside the values Arakon mentioned then this is not a bug/issue, it is a designed feature that allows the FC to go into failsafe when rx is out of range.

closed 2016-06-18 by KiteAnton