betaflight/betaflight #550: Betaflight 3.0.0 high CPU usage.

AndersHoglund, 2016-06-19:
Betaflight 3.0.0 high CPU usage.
I noticed very high CPU load with 3.0.0. on  Flip32F3 (SPRF3 clone) compared to 2.8.1.

With default settings (erased when flashing), CPU:13%
Adding only 250us/4kHZ : CPU:77%
Adding all my basic settings and features: CPU:100%

Doing exactly the same test with 2.8.1 I get CPU:6%, CPU:9% and CPU:11%. 

Something is not well with 3.0.0....
See attached files for more details. 
/A
[bf300_cpu_load.txt](https://github.com/betaflight/betaflight/files/322175/bf300_cpu_load.txt)
[basic_settings_Flip32F3.txt](https://github.com/betaflight/betaflight/files/322176/basic_settings_Flip32F3.txt)

blckmn, 2016-06-19:
Does it settle after a set amount of time (say 45seconds?)? or stay high?

There's no SD card on that one is there?

AndersHoglund, 2016-06-19:
It stays constantly high. No SD-card, onboard SPI-flash.
Both tests done on the bench with a bare board, only Spek-Sat connected. Does not seem to matter whether Tx is on or not.

AndersHoglund, 2016-06-19:
Same on a SPRF3 BG clone as well.

borisbstyle, 2016-06-19:
Can you check "tasks"  to see what task is responsible?

AndersHoglund, 2016-06-19:
Well, tasks are in the attatched "bf300_cpu_load.txt". 
The only thing I can see different in 281 vs 300 is the PID task,  ~140us vs ~250us.

borisbstyle, 2016-06-19:
@AndersHoglund 250 is not enough to run 4K. That generates high CPU load probably

AndersHoglund, 2016-06-19:
Yeah, 250 is really too much even on 2kHz.
Is it only me seeing this? Using my own Jenkins builds for test. Could someone confirm or reject this issue, using a build in another environment. Mind you, I used my Jenkins builds in both the 281 and 300 tests .

borisbstyle, 2016-06-19:
I will start testing/debugging soon.

AndersHoglund, 2016-06-21:
Re-checked CPU load after the latest 2.8.1 to 3.0.0 merge/re-base. Still high CPU load and slow PID task. Looks like this after flashing with full chip erase on a SPRF3 BG clone : 

    # version
    # BetaFlight/SPRACINGF3 3.0.0 Jun 22 2016 / 00:34:09 (5d385d0)
    # status
    System Uptime: 32 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:14%
    CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
    Cycle Time: 526, I2C Errors: 5, config size: 1292
    
    # tasks
    Task list:
    0 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 0ms
    1 - (PID) max: 327us, avg: 245us, rate: 950hz, total: 17sec
    - - (GYRO)  rate: 1901hz
    2 - (ACCEL) max: 242us, avg: 195us, rate: 967hz, total: 7sec
    3 - (ATTITUDE) max: 59us, avg: 28us, rate: 97hz, total: 184ms
    4 - (RX) max: 109us, avg: 78us, rate: 49hz, total: 170ms
    5 - (SERIAL) max: 23405us, avg: 2us, rate: 97hz, total: 100ms
    6 - (BATTERY) max: 11us, avg: 0us, rate: 48hz, total: 5ms
    7 - (BEEPER) max: 9us, avg: 0us, rate: 95hz, total: 5ms

PID task taking 245uS average is the worrying part I think. 4/2/2kHz max, 4/4/4 impossible. With 2.8.1 there is no problem running full 4/4/4 .

blckmn, 2016-06-22:
@AndersHoglund does the SPRF3 clone use VCP or have a usb serial chip on board?

AndersHoglund, 2016-06-22:
SPRF3 has the CP2102 USB-serial on UART1, no VCP,

savaga, 2016-06-22:
My dev board has similar issue with 2.8.1 @ 8khz/8khz:

```
# version
# BetaFlight/SIRINFPV 2.8.1 Jun 19 2016 / 19:09:05 (3828eb9)
# status
System Uptime: 63 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:100%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
Cycle Time: 209, I2C Errors: 0, config size: 1096

# tasks
Task list:
0 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 575us, avg: 122us, rate: 4901hz, total: 58sec
2 - (ACCEL) max: 42us, avg: 1us, rate: 319hz, total: 397ms
3 - (ATTITUDE) max: 59us, avg: 27us, rate: 80hz, total: 223ms
4 - (RX) max: 95us, avg: 63us, rate: 39hz, total: 191ms
5 - (SERIAL) max: 685us, avg: 2us, rate: 84hz, total: 85ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 30hz, total: 8ms
7 - (BEEPER) max: 7us, avg: 0us, rate: 45hz, total: 4ms
9 - (OSD) max: 1620us, avg: 121us, rate: 48hz, total: 229ms
```

But it seems ok with latest 3.0.0 @8khz/8khz:

```
# version
# BetaFlight/SIRINFPV 3.0.0 Jun 20 2016 / 23:05:46 (b8b798d)
# status
System Uptime: 133 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:43%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
Cycle Time: 130, I2C Errors: 0, config size: 1116

# tasks
Task list:
0 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 1ms
1 - (GYRO/PID) max: 584us, avg: 88us, rate: 7633hz, total: 94sec
2 - (ACCEL) max: 43us, avg: 1us, rate: 906hz, total: 2sec
3 - (ATTITUDE) max: 62us, avg: 29us, rate: 99hz, total: 650ms
4 - (RX) max: 95us, avg: 64us, rate: 49hz, total: 533ms
5 - (SERIAL) max: 638us, avg: 2us, rate: 99hz, total: 243ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 19ms
7 - (BEEPER) max: 11us, avg: 0us, rate: 98hz, total: 18ms
11 - (OSD) max: 1758us, avg: 61us, rate: 59hz, total: 579ms
```

KiteAnton, 2016-06-23:
I get the same issue on a SPRacingF3, 

```
 version
# BetaFlight/SPRACINGF3 2.8.1 Jun 19 2016 / 20:31:49 (0c5ea0f)
# status
System Uptime: 51 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:18%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 251, I2C Errors: 0, config size: 1292

# tasks
Task list:
0 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 239us, avg: 181us, rate: 3875hz, total: 39sec
4 - (RX) max: 115us, avg: 83us, rate: 49hz, total: 269ms
5 - (SERIAL) max: 16371us, avg: 2us, rate: 98hz, total: 51ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 8ms
7 - (BEEPER) max: 7us, avg: 0us, rate: 98hz, total: 8ms
14 - (TELEMETRY) max: 9us, avg: 0us, rate: 496hz, total: 116ms

# 

# version
# BetaFlight/SPRACINGF3 3.0.0 Jun 23 2016 / 22:22:59 (23e2fd5)
# status
System Uptime: 13 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 300, I2C Errors: 0, config size: 1296

# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 4hz, total: 0ms
1 - (GYRO/PID) max: 319us, avg: 286us, rate: 3311hz, total: 14sec
4 - (RX) max: 110us, avg: 79us, rate: 24hz, total: 36ms
5 - (SERIAL) max: 604us, avg: 2us, rate: 14hz, total: 9ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 7hz, total: 0ms
7 - (BEEPER) max: 8us, avg: 0us, rate: 14hz, total: 0ms
14 - (TELEMETRY) max: 0us, avg: 0us, rate: 65535hz, total: 0ms

```

AndersHoglund, 2016-06-23:
Thanks for the confirmation Anton. 
Betaflight 2.8.2 is still some 100us average faster with the PID task:

    # version
    # BetaFlight/SPRACINGF3 2.8.2 Jun 23 2016 / 02:10:24 (4c59769)
    # tasks
    ......
    1 - (PID) max: 226us, avg: 143us, rate: 998hz, total: 7sec
    ......

What does this look like on other F3 based FC's? I have only BG SPRF3, X-Racer v1 and Flip32F3. All SPRF3 clones, running the same code.

MJ666, 2016-06-23:
i did some testing here. The AlienFlight F4 test board with default settings shows normally 2% CPU usage. If there is no SDCard inserted the CPU usage goes to 7% for a while and is calming down to 2%. With inserted SDCard it is always 2%. With Gyro update frequency of 4kHz is will go to 7-8% but with 8kHz it goes to 100%. It also looks to be the MPU6500 is running with the lowest SPI bus Clock? This may make SPI reading much more slow.

resokk, 2016-06-23:
@MJ666 the same result on BlueJayF4

blckmn, 2016-06-23:
I'll up the SPI rate and submit a PR for testing

MJ666, 2016-06-24:
I did some testing with the actual SPI standard clock (11.5Mhz). This is working for me even with long wiring on my test environment. Here the CPU will be only 17% with the 8kHz gyro update. With 5.25MHz the CPU load is 19%.

blckmn, 2016-06-24:
Can we please recheck these targets now that the SPI speed has been updated :+1:

AndersHoglund, 2016-06-24:
BG SPRF3 seems unaffected, same as before. PID task taking 250us :

    # version
    # BetaFlight/SPRACINGF3 3.0.0 Jun 24 2016 / 16:54:29 (fefd34d)
    # tasks
    ....
    01 - (PID) max: 330us, avg: 248us, rate: 950hz, total: 29sec
    ....

Not that strange, as it does not have SPI connected gyro/acc.

borisbstyle, 2016-06-24:
you guys need to start profiling the code.

split the code sections in multiple sections and measure the timing to get
closer to the code section what decreased the performance.

blckmn, 2016-06-24:
SPRF3 incorrectly had VCP enabled. Not sure if it's a cause, but it's now disabled with latest merge. I'll start looking in the morning now I have a clone working.

AndersHoglund, 2016-06-24:
VCP has nothing todo with this issue. Checked the latest merge (fefd34) .

resokk, 2016-06-25:
On new version everything seems much better (BlueJay F4):
System Uptime: 96 seconds, Voltage: 159 * 0.1V (4S battery - OK), CPU:19%
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500
Cycle Time: 130, I2C Errors: 0, config size: 1096

00 - (SYSTEM) max: 1us, avg: 0us, rate: 9hz, total: 0ms
01 - (PID) max: 4547us, avg: 84us, rate: 3846hz, total: 113sec
 - - (GYRO)  rate: 7692hz
02 - (ACCEL) max: 18us, avg: 0us, rate: 990hz, total: 1sec
03 - (ATTITUDE) max: 25us, avg: 0us, rate: 99hz, total: 250ms
04 - (RX) max: 47us, avg: 13us, rate: 49hz, total: 292ms
05 - (SERIAL) max: 3399us, avg: 1us, rate: 98hz, total: 155ms
06 - (BATTERY) max: 40005us, avg: 0us, rate: 49hz, total: 91ms
07 - (BEEPER) max: 4us, avg: 0us, rate: 98hz, total: 8ms

gyro_sync_denom = 1
pid_process_denom = 2

borisbstyle, 2016-06-25:
My first board testing.
NAZE target. CPU usage is exactly the same like on 2.8. It simply runs fine!

borisbstyle, 2016-06-25:
Second Target:
Colibri Race: Keeps rebooting. Reboot cycle

borisbstyle, 2016-06-25:
It actually suddenly started to work now

borisbstyle, 2016-06-25:
Also COLIBRI_RACE target runs normal CPU! No difference with 2.8

AndersHoglund, 2016-06-25:
Well, it all depends on how you define "normal" and "runs fine".... 
Did a quick check on my NAze32 and there is a 100us difference in the PID tasks here too between 2.9.0 and 3.0.0, average 240us vs 335us

    # version
    # BetaFlight/NAZE 3.0.0 Jun 25 2016 / 22:15:52 (bbdf593)
    # status
    System Uptime: 17 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:4%
    CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
    Cycle Time: 1008, I2C Errors: 0, config size: 1280
    
    # tasks
    Task list:
    00 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
    01 - (PID) max: 591us, avg: 335us, rate: 498hz, total: 6sec
    - - (GYRO)  rate: 997hz
    02 - (ACCEL) max: 470us, avg: 239us, rate: 332hz, total: 1sec
    03 - (ATTITUDE) max: 323us, avg: 288us, rate: 99hz, total: 576ms  
    04 - (RX) max: 120us, avg: 85us, rate: 49hz, total: 102ms
    05 - (SERIAL) max: 602us, avg: 2us, rate: 99hz, total: 26ms
    06 - (BATTERY) max: 78us, avg: 1us, rate: 49hz, total: 6ms
    07 - (BEEPER) max: 9us, avg: 0us, rate: 99hz, total: 2ms

    # version
    # BetaFlight/NAZE 2.9.0 Jun 25 2016 / 23:43:09 (3d9b180)
    # status
    System Uptime: 14 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:3%
    CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
    Cycle Time: 1007, I2C Errors: 0, config size: 1280
    
    # tasks
    Task list:
    0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
    1 - (PID) max: 484us, avg: 240us, rate: 499hz, total: 4sec
    - - (GYRO)  rate: 999hz
    2 - (ACCEL) max: 358us, avg: 139us, rate: 333hz, total: 1sec
    3 - (ATTITUDE) max: 301us, avg: 267us, rate: 99hz, total: 524ms
    4 - (RX) max: 116us, avg: 85us, rate: 49hz, total: 96ms
    5 - (SERIAL) max: 624us, avg: 2us, rate: 99hz, total: 32ms
    6 - (BATTERY) max: 74us, avg: 1us, rate: 49hz, total: 6ms
    7 - (BEEPER) max: 8us, avg: 0us, rate: 99hz, total: 3ms

borisbstyle, 2016-06-26:
@AndersHoglund 
How come I didnt see the difference?

Let me check again on the naze

borisbstyle, 2016-06-26:
3.0
```
# version
# BetaFlight/NAZE 3.0.0 Jun 26 2016 / 00:17:02 (8b997da)
# tasks
Task list:
00 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
01 - (GYRO/PID) max: 477us, avg: 315us, rate: 2624hz, total: 20sec
04 - (RX) max: 119us, avg: 88us, rate: 49hz, total: 118ms
05 - (SERIAL) max: 11156us, avg: 2us, rate: 97hz, total: 31ms
06 - (BATTERY) max: 75us, avg: 1us, rate: 49hz, total: 7ms
07 - (BEEPER) max: 9us, avg: 0us, rate: 97hz, total: 3ms

# status
System Uptime: 27 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:23%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 378, I2C Errors: 0, config size: 1280
```

borisbstyle, 2016-06-26:
2.9

```
# version
# BetaFlight/NAZE 2.9.0 Jun 25 2016 / 23:43:09 (3d9b180)
# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 480us, avg: 319us, rate: 2638hz, total: 10sec
4 - (RX) max: 116us, avg: 84us, rate: 49hz, total: 60ms
5 - (SERIAL) max: 337us, avg: 2us, rate: 97hz, total: 18ms
6 - (BATTERY) max: 75us, avg: 1us, rate: 49hz, total: 4ms
7 - (BEEPER) max: 7us, avg: 0us, rate: 97hz, total: 1ms

# status
System Uptime: 14 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:23%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 374, I2C Errors: 0, config size: 1280
```

borisbstyle, 2016-06-26:
Accelerometer was disabled and Gyro/Pid set to 2.6khz in this case

borisbstyle, 2016-06-26:
Using 4.9 2015 ARM compiler

borisbstyle, 2016-06-26:
I also tried with ACC enabled etc and I get exactly same results on any settings

borisbstyle, 2016-06-26:
3.0

```
# version
# BetaFlight/NAZE 3.0.0 Jun 26 2016 / 00:17:02 (8b997da)
# tasks
Task list:
00 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
01 - (GYRO/PID) max: 484us, avg: 318us, rate: 989hz, total: 3sec
02 - (ACCEL) max: 361us, avg: 140us, rate: 333hz, total: 643ms
03 - (ATTITUDE) max: 314us, avg: 282us, rate: 99hz, total: 350ms
04 - (RX) max: 120us, avg: 88us, rate: 49hz, total: 63ms
05 - (SERIAL) max: 477us, avg: 2us, rate: 99hz, total: 20ms
06 - (BATTERY) max: 75us, avg: 1us, rate: 49hz, total: 3ms
07 - (BEEPER) max: 9us, avg: 0us, rate: 99hz, total: 1ms

# status
System Uptime: 13 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:4%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Cycle Time: 1009, I2C Errors: 0, config size: 1280
```

2.9
```
# version
# BetaFlight/NAZE 2.9.0 Jun 25 2016 / 23:43:09 (3d9b180)
# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 486us, avg: 318us, rate: 994hz, total: 3sec
2 - (ACCEL) max: 356us, avg: 139us, rate: 312hz, total: 545ms
3 - (ATTITUDE) max: 310us, avg: 279us, rate: 98hz, total: 296ms
4 - (RX) max: 116us, avg: 84us, rate: 49hz, total: 52ms
5 - (SERIAL) max: 490us, avg: 2us, rate: 98hz, total: 20ms
6 - (BATTERY) max: 75us, avg: 1us, rate: 49hz, total: 3ms
7 - (BEEPER) max: 8us, avg: 0us, rate: 99hz, total: 1ms

# status
System Uptime: 12 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:3%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Cycle Time: 1008, I2C Errors: 0, config size: 1280
```

borisbstyle, 2016-06-26:
I tried a huge list of targets and none of them have performance decrease! In fact there is a sligh performance increase.

So guys what are we looking at? Compiler version? I know there was an issue with newer compilers with VCP

blckmn, 2016-06-26:
VCP fixes are in merged this morning. I think the issue was SPI and i2c speed. Which was resolved this morning (Aust time). So it might just be a case of some needing to pull the latest code.

KiteAnton, 2016-06-26:
Just tried this again on a SPRacingF3 in latest betaflight branch and the issue seems to be fixed, CPU is "normal" again. 

```
# version
# BetaFlight/SPRACINGF3 3.0.0 Jun 26 2016 / 09:15:57 (db5453b)
# status
System Uptime: 58 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:13%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 257, I2C Errors: 0, config size: 1296

# tasks
Task list:
00 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 0ms
01 - (GYRO/PID) max: 232us, avg: 175us, rate: 3984hz, total: 45sec
04 - (RX) max: 103us, avg: 72us, rate: 49hz, total: 311ms
05 - (SERIAL) max: 591us, avg: 2us, rate: 98hz, total: 231ms
06 - (BATTERY) max: 12us, avg: 0us, rate: 49hz, total: 10ms
07 - (BEEPER) max: 8us, avg: 0us, rate: 98hz, total: 10ms
```

borisbstyle, 2016-06-26:
Great

We can continue further with flight testing.

I am also curious to different features. PWM, SBUS, LEDs etc.

borisbstyle, 2016-06-26:
After we verified stuff I will start announcing beta releases to the public
to try out.

AndersHoglund, 2016-06-26:
OK, but with a warning. Max 4/2/2kHz on F3's . 
A bit of a downgrade it will be.

borisbstyle, 2016-06-26:
f3 spi runs fine 8/8

AndersHoglund, 2016-06-26:
Yepp, all is well on the development branch. There was some SPI and I2C commits I beleive. 
Good job guys. Case closed.

For the record:

    # version
    # BetaFlight/SPRACINGF3 3.0.0 Jun 26 2016 / 04:20:07 (eb55c91)
    # status  
    System Uptime: 28 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:6%
    CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
    Cycle Time: 504, I2C Errors: 5, config size: 1296
    
    # tasks
    Task list:
    00 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
    01 - (PID) max: 218us, avg: 139us, rate: 994hz, total: 9sec
    - - (GYRO)  rate: 1988hz
    02 - (ACCEL) max: 132us, avg: 86us, rate: 990hz, total: 3sec
    03 - (ATTITUDE) max: 57us, avg: 26us, rate: 99hz, total: 164ms
    04 - (RX) max: 108us, avg: 77us, rate: 49hz, total: 149ms
    05 - (SERIAL) max: 634us, avg: 2us, rate: 99hz, total: 40ms
    06 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 4ms
    07 - (BEEPER) max: 9us, avg: 0us, rate: 99hz, total: 4ms

closed 2016-06-26 by AndersHoglund