betaflight/betaflight #553: Save issues in "Configuration" tab

Wesstone, 2016-06-19:
I just flashed my rev3a Dodo with the 2.8.1 Spracing hex.  Im using an XSR sbus receiver with smart port telemetry.
In the "Ports" tab...
I selected smart port on UART2 and turned Serial RX on for UART3 then saved.
Everything saved fine.
Then I go into the "Configuration" tab, select Serial RX and Sbus, turn telemetry on, set my min/max throttle and save.  Except the only thing that saves are the min/max throttle.  Everything else reverts back to default and Oneshot125 is defaulted off.
Ive tried this numerous times.  Erased and reflashed firmware with the same results.
I reverted to 2.7.1 and everything works as it should.

arrowcircle, 2016-06-19:
Same problem with clone SP F3.
Flashed 2.8.1 Release, got problem with configuration tab save.
Reverted back to 2.8.1 RC1 version and all is ok

bluesign2k, 2016-06-19:
+1 for similar issues with most items reverting back after clicking save on config tab. I've found that settings can be changed and saved when using CLI. I also find that I can change the looptime to 500 us and save succusfully (checked by unplugging and re-plugging and checking settings), then unplug again and go fly, then plug in again to download blackbox after flight and the looptime is back to 365 us - I've seen this a few times but I can't consistently re-create the problem. Using SPRF3.

borisbstyle, 2016-06-19:
Re-uploading 2.8.1 with the fix in 5minutes

closed 2016-06-19 by borisbstyle