betaflight/betaflight #563: Problems with lates PWM changes and brushed FC's

MJ666, 2016-06-23:
I have seen some strange behavior since latest pwm changes last days (adding the use sync pwm output back). If I change the mixer the brushed AlienFlight controller activate all the motors and it will go in infinite loop always trying to start motors and will not finish the configuration. USB power will shutoff and he cycle will start again. Checking the config afterwards in CLI the motor_pwm_protocoll is empty and if an motor is connected it will run with speed. With disconnected motors the change will go without problems.

This is may be also related to issues user reporting that motors starting up on brushless builds when they do configuration changes and have battery connected.

blckmn, 2016-06-24:
Thanks @MJ666 I'll have a look tonight.

blckmn, 2016-06-24:
@MJ666 https://github.com/betaflight/betaflight/issues/438 highlights the issue in 2.8.x also.

I have a suspicion that the issue will only occur the first time you flash, and you need to reset settings or save settings with battery disconnected at least once.

MJ666, 2016-06-24:
#438 was related to wrong default settings for the AlienFlight targets and was fixed. The problem now happens with the brushed copters also with no battery connected. Motors will be powernd by USB also but the current draw will make the USB port to switch off and the FC can not strart properly untill all motors get disconnected. it will affect all of the brushed FC's and many users even solder motors to the FC to save weight. There was no problem in 3.0 until the unsycnced pwm parameter was reintroduced some days ago.

blckmn, 2016-06-24:
Hey @MJ666 does this problem occur if full chip erase is done on flashing?

MJ666, 2016-06-24:
will not happen during flashing and resetting the config. It only happens when the mixer is changed.

blckmn, 2016-06-24:
Aha. Ok thanks. Will check now.

blckmn, 2016-06-24:
What mixer do you change to?

MJ666, 2016-06-24:
To "Hex X" have not tried anything else.

blckmn, 2016-06-25:
Hi @MJ666 looking at the scope after setting brushed. There is always a little pulse on save. I don't need to change the mixer to get it. The pulse occurs after save though, and on reboot. I get the pulse if pulling NRST down and rebooting the FC - its only tiny - but it could cause a problem. Do you think this could be the issue? or am I looking at it wrong.

MJ666, 2016-06-25:
Some very short pulses should not be an problem. But if the FET's get active longer the motors will start and USB power will cut off. Ideally we should not have anything on the outputs except the signal defined by the motor_pwm_protocol. I checked the useunsync motor output code and this may also influence the timers. Likely it should be on for brushed motors. I need to do some new test here.

MJ666, 2016-06-25:
Have done some mor testing. Looks to be after changing the mixer the motor_pwm_protocol seting is messed up:

this is how it looks at the beginning:
`# set pwm
use_unsynced_pwm = OFF
Allowed values: OFF, ON

motor_pwm_protocol = BRUSHED
Allowed values: OFF, ONESHOT125, ONESHOT42, MULTISHOT, BRUSHED

motor_pwm_rate = 32000
Allowed range: 200 - 32000

servo_pwm_rate = 50
Allowed range: 50 - 498
`
an this is after the mixer change.
`# set pwm
use_unsynced_pwm = OFF
Allowed values: OFF, ON

motor_pwm_protocol = #ÿÿ

ÿ2ÿ2ÿÿÿÿ

((
(
((

Fÿ22ÿ


ÿ
Allowed values: OFF, ONESHOT125, ONESHOT42, MULTISHOT, BRUSHED

motor_pwm_rate = 32000
Allowed range: 200 - 32000

servo_pwm_rate = 50
Allowed range: 50 - 498

#`

blckmn, 2016-06-27:
Hey @MJ666 i think this is sorted now yeah, with your changes to the configurator?

MJ666, 2016-06-27:
Yes, problem solved. Closing.

closed 2016-06-27 by MJ666