betaflight/betaflight #565: Looptime scaling in the PID loop

rav-rav, 2016-06-23:
There will always be some jitter in time between gyro loops and consequently PID loops. Sometimes there will be a little more time between updates and sometimes less.
Calculations in the PID loop are supposed to compensate for looptimes, but don't take into account this jitter.

The function used to obtain dT always returns the same constant value. Is there a reason why this is used instead of the real time delta?
A large bump in the gyro signal caused by a delayed readout would be compensated by the real time delta and result in smoother measurements.

```
float getdT (void) {
    static float dT;
    if (!dT) dT = (float)targetPidLooptime * 0.000001f;

    return dT;
}
```

borisbstyle, 2016-06-23:
@rav-rav 
Actually this is minimizing jitter. Making dt adoptable to jitter just gives more jitter! It is really just meant to adopt the same PID's to different looptimes.

When you send command to ESC it can take between 0 and 128us before that one starts applying it to the motors. Also all calculations are based on the latest gyro sample, which is always on time! Even if you receive gyro sample later it was still generated on the correct time.

One of the first things I did in betaflight was to get rid of this looptime adoption, which drastically reduced the jitter in the motors

rav-rav, 2016-06-23:
> Even if you receive gyro sample later it was still generated on the correct time.
Yes, then it makes sense to use a constant value. I was not sure if this was the case or if the actual time when the sample is taken was changing. Thanks for explaining.

closed 2016-06-23 by rav-rav