betaflight/betaflight #571: CHANNEL_FORWARDING not working in SP3Racing FC

Nandox7, 2016-06-23:
Hi,

Enabling this feature in a SP3Racing FC doesn't forward the AUX[1-4].
Tested using several Betaflight versions including the current one 2.8.1 in QuadX mixer mode.

The expected behaviour would be that channels Aux[1-4] would be forwarded to Motor/PWM pins 5 to 8.

(Tested the same setup on a Naze32Rev5 and it worked as expected.)

borisbstyle, 2016-06-23:
did you enable aux channels with max_aux_channels?

Nandox7, 2016-06-23:
No I have not.
I followed the same process as I was doing previously in the Naze32 that is just enabled the feature in the UI.

rav-rav, 2016-06-24:
You need to enable FEATURE_SERVO_TILT to enable output to other channels.
Then you can use CHANNEL_FORWARDING to forward your inputs.

This really should be reworked.

Nandox7, 2016-06-24:
I may be wrong but at some stage i tried that as well.
Well i tried a lot of things. :)
Smix, servo, ...

I started looking at the code and i wonder if there isnt some conflict in
the conditions checking.
There are specific rules set for the SP3Racing.

rav-rav, 2016-06-24:
I have a SP3Racing FC and use channel forwarding to activate a lost plane beeper on my PDB via a switch on my TX. Works fine, but I'd have to check the exact settings. I don't have access to my quad right now however.

Nandox7, 2016-06-24:
Would appreciate that.
I need it for the same purpose, i use a Matek 5-in-1 that has buzzer
builtin but needs a PWM input to control it.

rav-rav, 2016-06-24:
I think also servo 1 and 2 are bound to pin 7 and 8. 5 and 6 can not be used.
Did you try those?

Nandox7, 2016-06-24:
I have it connected to Motor pin 8 so that would work with AUX4.
At some stage i tried and set rules for all servos using smix and later the
servo command.
For last i plugged a servo from M5 to M8....and had no luck.
But i may have tried to move M5 with Aux1, M6 with Aux2 and so on...
I'll try it again.

rav-rav, 2016-06-24:
I'm using AUX4 to control the beeper. It is connected to motor pin 7, will beep at 1000 and does not like signals over 1500.

Here's what I'm using:
`
feature SERVO_TILT
feature CHANNEL_FORWARDING
servo 0 1000 1500 1500 90 90 100 7
`
The last line means that AUX4 is used as input and a min value of 1000 and a max value of 1500 is forwarded to servo 0.
Servo 0 means motor pin 7.
Servo 1 is motor pin 8.

Nandox7, 2016-06-24:
Bingo. Just changed to use M8 instead and it is working now.
Thank you!

I'd still say the feature is not working as it should but at least this
works as a decent
workaround in the mean time.

-Nando

KiteAnton, 2016-07-06:
Issue seems resolved, closing.

closed 2016-07-06 by KiteAnton