betaflight/betaflight #572: Cannot set "telemetry_inversion" to Off state.

laurynas-karvelis, 2016-06-23:
Hi all,

Using Betaflight 2.8.1, I've got this odd situation on the XRacer 303 v2.1 board (F3 chip) with 3 UART ports available. My connections on the board: USB and S.Port Telemetry on UART1, UART2 - OSD TX/RX, UART3 - Sbus. The problem, as I `set telemetry_inversion = Off on console and then `save`, once the board is restarted, it reverts `telemetry_inversion` param back to `On` state.

Is this a bug? As right now, all I get, is the basic telemetry data, like receiver voltage, RSSI, but no main battery voltage and other params I used to get on my Taranis. VBAT is connected.

Thank you so much in advance for any feedback!

laurynas-karvelis, 2016-06-23:
Just figured it out. Went to read the source code inside [smartport.c](https://github.com/betaflight/betaflight/blob/betaflight/src/main/telemetry/smartport.c#L141), and what I've found that SmartPort needs 57600 baud.

Now UART1 configuration by default is set with MSP 115200 baud, and Telemetry column defaults baud to AUTO. What I think is happening, once the board detects that it's not connected to USB, it just leaves 115200 baud as is, so the SmartPort cannot talk properly as UART1 isn't on 57600 baud.

Hope this makes sense.

tannewt, 2016-06-23:
So, do you actually need inversion off? Its impossible to turn off with the F3s because of this code: https://github.com/betaflight/betaflight/blob/betaflight/src/main/config/config.c#L891

Sounds like you actually have another issue with the UART settings. Did you arm (with props off) to test it? I'm not exactly sure how the serial port sharing code works.

laurynas-karvelis, 2016-06-23:
@tannewt, you are right, I didn't have to turn telemetry_inversion off completely. My only problem was that if USB and SmartPort being used via UART1, then, not only MSP needs to be enabled on it, but on the Telemetry column, choosing SmartPort you also need to pick 57600 baud too. That's the important part.

Once the board is restarted and armed, you can now see telemetry data with VBAT and other indicators.

KiteAnton, 2016-06-23:
@laurynas-karvelis, good finding. Please close this issue if you consider it solved.

closed 2016-06-23 by laurynas-karvelis