betaflight/betaflight #575: naze32 flyskyi6a pwm

xeroorc, 2016-06-24:
using betaflight on naze32 but unable to even throttle my reciever  and motors. connection all checked and should be correct. 
all bars not moving. please help

KiteAnton, 2016-07-06:
When posting issues at least post the following info. 
`status`
`tasks`
`dump`

And also describe your problem more. You are not mentioning your HW setup and what you have done to test it. 

Update the issue with additional info or issue will be closed.

dako76, 2016-07-07:
Hello same issu here, have to set my esc to PWM, but with this new firmware my quad is not flyable tried to tune my PID but P is always to high or to low best flyable was at P 13 for all and reduced I and D to 5 anyone can help me with this? Otherwise i have to go back to cleanflight this is working much more well!

borisbstyle, 2016-07-09:
Closing due to starvation of discussion without detailled info

closed 2016-07-09 by borisbstyle